olikraus/u8g2 #1038: OLED(128x128/1.5") Stuttering / cropped text

FALL1N1, 2019-11-04:
Hello

I'm not entirely sure if the issue is within my device/display or the library, but some of the fonts are rendered properly while others not.

Here is a sample video - https://www.youtube.com/watch?v=E9AwZ_WyeKA 

The wiring is correct and the code is the basic from the examples:

`#include <Arduino.h>
#include <SPI.h>
#include <U8x8lib.h>

U8X8_SSD1327_WS_128X128_4W_SW_SPI u8x8(4, 3, 6, 5, 2);

void setup(void) {
  u8x8.begin();
  u8x8.setFont(u8x8_font_8x13B_1x2_f);
}

void loop(void) {
  u8x8.drawString(0,0,"XxXxXxXxXxXxXxXxXx");
  delay(1000);
}`

Any ideas would be appreciated.

olikraus, 2019-11-04:
I am a little bit surprised. I never saw such artefacts on the screen.
One issue could be: If the number of chars exceed the number of chars per screen, then the result is unpredictable.

What kind of board did you use?

FALL1N1, 2019-11-05:
Sorry for the late reply. It is one of these chinese replicas as far as I know, example: https://www.ebay.com/p/1884261551?iid=273894356458&rt=nc

Although the display works well with it's own library so my guess is that the chip is too slow for u8g2

olikraus, 2019-11-05:
Did you use level shifters? Uno is a 5v controller, ssd1327 is a 3.3v oled

FALL1N1, 2019-11-06:
It has two pins for 3.3v output, I use them, although that specific waveshare oled does work with both 3.3v/5v, https://www.waveshare.com/1.5inch-oled-module.htm

olikraus, 2019-11-06:
it was just an idea...

FALL1N1, 2019-11-06:
I was just providing information, sorry if it was misunderstood. Although, managed to fix it by using the library on a genuine Arduino board, worked like a charm so most eventually the issue was with the cheaper board.

olikraus, 2019-11-06:
Sounds good :)

FALL1N1, 2019-11-14:
Update: I have replaced the board with Iteaduino Leonardo with a 3.3v/5v switch, works like a charm while the board itself is working on 3.3v, so my guess is that the cheap chinese boards are 5v and it does not work, so I am now closing this issue since it's invalid.

Thank you for your time!

closed 2019-11-14 by FALL1N1

olikraus, 2019-11-14:
Good to see you solved the problem. Thanks for the feedback.