olikraus/u8g2 #1039: Compatibility with Parallel 8080 on Linux

Mohammad1098, 2019-11-04:
Hi , Thanks for the great and huge library.

I working on display that has SSD1309 Driver. I came up with a local library that does the basic commands to communicate with SSD1309 using Parallel port on linux based on 8080 mode.

The steps to display a string on display is too hard and complicated , so could you please explain how did you reduce the hard work to display string ? what are the components that doing the display staff then I can adopted with my code

olikraus, 2019-11-04:
> The steps to display a string on display is too hard and complicated , so could you please explain how did you reduce the hard work to display string ?

I did not reduce something. It was **hard** and **long** work and it took many many hours.

> what are the components that doing the display staff then I can adopted with my code

Components are [bdfconv](https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv) and the [u8g2 font render](https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c) subsystem.

Mohammad1098, 2019-11-05:
could you please explain the process when the drawStr() func is called in high level and what is the relation with bdfconv and font render 

Thanks

olikraus, 2019-11-05:
"drawStr()" calls the internal function "u8g2_draw_string()". "u8g2_draw_string()" handles the string rotation and the unicode or ASCII decoding (depending on the setup in "drawStr()".

"u8g2_draw_string()" will then call "u8g2_DrawGlyph()" for each decoded glyph.
"u8g2_DrawGlyph" will apply the vertical adjustment and then calls "u8g2_font_draw_glyph" for the glyph (encoding).

"u8g2_font_draw_glyph" will get the glyph bitmap data, which was previously generated by "bdfconv" and decode the bitmap data with "u8g2_font_decode_glyph"

Mohammad1098, 2019-11-07:
if I'm using parallel port 0x378 which is contains pins d0 to d7 on Linux and I would like to use the library is it possible? or I have to edit lots of files to make the library compatible with Linux environment?

olikraus, 2019-11-07:
No, you just have to create a gpio and delay callback for your PC:

https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

Mohammad1098, 2019-11-07:
> No, you just have to create a gpio and delay callback for your PC:
> 
> https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

I'm confused. what is the purpose of the third parameter "Communication Callback" and the fourth parameter "GPIO AND DELAY " 

why we did not define GPIO in the communication callback ?

-are they depend on each other ? 
-should I implement both of them ?

thank you for your time

olikraus, 2019-11-07:
> I'm confused. what is the purpose of the third parameter "Communication Callback" 

Please see the instructions. Communication callback will be "u8x8_byte_8bit_8080mode"
See here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#communication-callback-eg-u8x8_byte_hw_i2c

> why we did not define GPIO in the communication callback ?

It allows you to switch between 8080 mode and 6800 mode without changing GPIO. You define the GPIOs once and then you just exchange the communication callback. Isn't this great?

> are they depend on each other ?

To some extend yes...

> should I implement both of them ?

Of course no, "u8x8_byte_8bit_8080mode" already exists... you just need to wrote a GPIO callback.
More specific, you do not need to implement everything in the GPIO callback, only those ports which are relevant for 8080... 
However, also the millisecond delay needs to be implemented.

Mohammad1098, 2019-11-09:
thank you for clarify.

last question , in constructor we need to pass communication callback (3rd) argument the and GPIO and Delay callback (4th) argument 

in both cases we have to pass msg. do we need to pass new message in every time we send data to display or what?

Thank you

olikraus, 2019-11-10:
The communication callback (3rd arg) is already defined. You must use one of the existing callbacks.
The 4th callback you have to provide.

> in both cases we have to pass msg. 

The msg are also defined by u8g2. You have to react on it. The messages are part of the template:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#template-for-the-gpio-and-delay-callback

Mohammad1098, 2019-11-10:
does this mean create a new constructor every time ?

or we just need to pass U8X8_MSG_BYTE_INIT in 3rd arg and U8X8_MSG_GPIO_AND_DELAY_INIT in 4th arg and other messages will be used by other functions

olikraus, 2019-11-10:
> does this mean create a new constructor every time ?

I really have no idea what you mean? Of course "no".

> or we just need to pass U8X8_MSG_BYTE_INIT in 3rd arg and U8X8_MSG_GPIO_AND_DELAY_INIT in 4th arg and other messages will be used by other functions

You have to pass the callback function. You should take the template code, rename it and fill in the code after the case statements. Then pass the renamed function name as argument to the constructor.

Mohammad1098, 2019-11-10:
> > does this mean create a new constructor every time ?
> 
> I really have no idea what you mean? Of course "no".
> 

I meant if there are many cases in each callback , this means we need to pass U8X8_MSG_BYTE_INIT message in 3rd arg and U8X8_MSG_GPIO_AND_DELAY_INIT message in 4th arg at the beginning of program and other cases will be used by other functions is this right ?

olikraus, 2019-11-10:
Again no: You must pass a function as 3rd and 4th argument.

See here for example: 
https://github.com/olikraus/u8g2/blob/9d61f30e7abd8987837fb2c09fb56391800554c7/sys/arm/lpc824/u8g2_logo/main.c#L87

Mohammad1098, 2019-11-10:
> Again no: You must pass a function as 3rd and 4th argument.
> 
> See here for example:
> https://github.com/olikraus/u8g2/blob/9d61f30e7abd8987837fb2c09fb56391800554c7/sys/arm/lpc824/u8g2_logo/main.c#L87

I agree with you but when I used 8080mode constructor I have to pass u8x8_t *u8x8 ,  uint8_t msg, uint8_t arg_int, void *arg_ptr otherwise an error occurred

thank you for your time

olikraus, 2019-11-10:
for 8080 mode you have to pass `u8x8_byte_8bit_8080mode` as mentioned here: 
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#communication-callback-eg-u8x8_byte_hw_i2c

closed 2019-11-23 by olikraus