olikraus/u8g2 #1041: HX1230 to use AVR hardware SPI (9-bit messages over 8-bit SPI subsystem)

spase74, 2019-11-08:
HX1230 is by default configured to use 3-wire SPI (do d/c pin) and accepts 9-bit messages. AVR microcontrollers cannot do it via their hardware SPI as it is limited to 8-bit messages only. One of possible solutions is send 9-bit messages through 8-bit SPI subsystem.
Attached is an example how to do it for u8x8 and u8g2. It works about 12-14ms faster than software SPI implementation (measured on Atmega32u4, 8mHz).

[helloworld.zip](https://github.com/olikraus/u8g2/files/3824357/helloworld.zip)

spase74, 2019-11-08:
Some comments how it works.
The library creates a 9-byte buffer to keep 8 9-bit messages. After that, each time when a message to the display is being sent, it is first converted to 9-bit (d/c bit (stored separately) as MSB and the rest after), shifted appropriately to keep bit-sequence and stored to the buffer. When the buffer becomes full it is sent to the display via hardware SPI.
Of course when we decoded to stop transfer, the buffer is not obligatory aligned to byte boundary, so we will send one garbage uncompleted message consisting of 1 to 8 bits. But then we'll rise CS and thus terminate sending the garbage - and this, according the display's datasheet, is ignored.

olikraus, 2019-11-08:
Thanks for the code.
I need to look into it...
But yes, it looks like a good idea....

olikraus, 2020-01-26:
I think we can fill the incomplete buffer with 0 bytes. This is the set column lo command, which will not harm the functionality.

https://github.com/olikraus/u8g2/issues/469

olikraus, 2020-01-26:
The 0 value can also be used for SSD1306

olikraus, 2020-01-26:
ToDo: Update codebuild

spase74, 2020-01-27:
Technically, no need to fill the buffer with zeros. The only issue produced by sending 9-bit words via 8-bit subsystem is that the last sent *byte* in common case contains a tail of the last *9-bit-word* and 1 to 8 bits of trash. But it is ok; we will raise CS and thus abort sending and this trash will be ignored by receiving controller.
I only mean; we do not need do bit-bangling for the last byte in the buffer; ok to send 1-8 bits of trash.
I use this solution in my "Open Skydive Altimeter" project, works perfectly.

olikraus, 2020-01-27:
I have tested the new code with a SSD1306 display:
```
  U8G2_SSD1306_128X64_NONAME_1_3W_SW_SPI	Uno			Clip=23.9 Box=41.6  @=4.3 Pix=7.5
  U8G2_SSD1306_128X64_NONAME_1_3W_HW_SPI	Uno			Clip=28.4 Box=58.4  @=4.4 Pix=7.9
```
3W HW SPI is probably 20% faster than SW SPI.

so... I think it works as expected. Thanks for the issue and the related code :-)

closed 2020-01-27 by olikraus