olikraus/u8g2 #1042: LX-12864B11

JeromeAr, 2019-11-08:
Hi. I have trouble to make the lib working with LX-12864B11 module from banggood (https://www.banggood.com/2_4-Inch-128x64-12864-Dot-Matrix-COG-LCD-Display-Module-With-Blue-Backlight-p-1403302.html ) 
The data says it s a ST7567 driver IC so I went for the U8X8_ST7567_JLX12864_4W_HW_SPI constructor in the various examples. I also tried several other configuration : HW and SW SPI and other LCD model with the same IC without success. I tryed both on teensy and stm32 bluepill always with the same issue : SPI seems to work well when probed with an oscilloscope but I only get a bunch of collunms flickering in random manner. Does anyone confront the same issue ?

olikraus, 2019-11-08:
Any pictures of the LCD with that flickering content possible?

olikraus, 2019-11-08:
A picture of the complete setup would be good too.

JeromeAr, 2019-11-08:
[image]
this is what I get with the 16x16 font demo with U8X8_ST7567_JLX12864_4W_HW_SPI.
the result is quite similar with SW version. the pattern is cyclic with the 2s period of the loop in the demo code. I can try to capture a video if you want.

JeromeAr, 2019-11-08:
the video : https://1drv.ms/v/s!Aj--d-HGQqVRiDL0oWW0WkQri4CZ?e=aie4zL

olikraus, 2019-11-08:
Can you make a picture of the flex cable connection?
Also: is there any datasheet for the display?

JeromeAr, 2019-11-08:
I have soldered wires directly on the flex cable pins. here s the datasheet. completed on the vendor page by an ugly screenshot.  
[image]
 . 
[LX-12864B11_V10 (1).pdf](https://github.com/olikraus/u8g2/files/3826073/LX-12864B11_V10.1.pdf)

JeromeAr, 2019-11-08:
I have tryed to modify the u8x8_st7567_jlx12864_init_seq[] in the u8x8_d_st7567.c file without success (I fist ha slowed down SPI speed, suspecting a trashy chinese clone of st7567 but without success neither

olikraus, 2019-11-08:
Thanks for all the answers. I am now back at my desk (i had mobile access only during the last view hours). This is what I think:
It is not really clear what is the root cause of the problems, however this *might* be the causes, started from the once which I think are the most probable ones:
1. The two caps are too far away from the connector. They should be connected directly near or at the flex cable.
2. There might be a power issue, especially because the power supply lines are also very long: Put a 100nF or a 1uF or both between pin 6 and 7 directly at the connector 
3. Protoboards are not reliable. Try to get rid of them. If all the caps are soldered near to the flex cable, then maybe you can at least avoid one of the protoboards. At least try to change the pins and check whether something changes (if protoboards would be reliable, the nothing will change)
If you have proper measure equipment you could also check the signals at the flex cable. Do clock, data, a0 and cs signals appear correctly at the LCD?
4. There might be a software issue. Indeed you could check multiple devices (the dimension  does not need to fit) starting from here https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7567-pi_132x64:
U8G2_ST7567_PI_132X64_1_4W_SW_SPI
U8G2_ST7567_JLX12864_1_4W_SW_SPI
U8G2_ST7567_ENH_DG128064_1_4W_SW_SPI
U8G2_ST7567_ENH_DG128064I_1_4W_SW_SPI
U8G2_ST7567_64X32_1_4W_SW_SPI
5. Can we trust the datasheet, that this is a ST7567? Maybe you could at least once try a ST7565 constructor... just to ensure that the chip isn' the problem...

(of course you could als use the HW SPI, i just took one constructor per device)

ok, these are the ideas from my side. Hope one of the points will help.

JeromeAr, 2019-11-09:
Thanks for all the help. I ll check this (already tried few of the points like using ST7565 or validating the signals at the connector with my scope). It's also possible that I got a defective unit, unfortunately I have bought only one display so I cannot run comparative tests. About the datasheet I am bit skeptical too and strangely, nobody has posted a picture of a working unit in commentary section of the commercial website. I think I ll order some more mainstream displays for my next projects ;-). I ll let you know how it goes. Anyway congrats for your work, it s a nice piece of software you are sharing.  Cheers !

closed 2019-11-09 by JeromeAr

olikraus, 2019-11-09:
I had a look at your video again, indeed the LCD might be broken, but it also just could be a power supply issue as mentioned above.