olikraus/u8g2 #1043: u8g2 not working with SPI nokia 5110 on esp8266

Andrew-Aitchison100, 2019-11-09:
Hi, I did a test using the "hello world" example code using an esp8266 node mcu board and a spi 1.3" OLED screen, and it worked perfectly.
       Then changing nothing but the set up line for the nokia 5110 screen, i re-uploaded the code, swapped the screen, and nothing was displayed.
     I know the nokia screen works as i tested it on an arduino nano. I cant comprehend what the problem could be. Does anyone here have an idea?

olikraus, 2019-11-10:
As far as I remember the nokia 5110 is a 5V display. I might not work with the ESP8266 unless you use level shifters on the data lines.

Andrew-Aitchison100, 2019-11-10:
I've seen multiple projects running the nokia 5110 straight off 3.3v from an esp8266. And a video tutorial goes over a data sheet specifying 3.3v. I'll keep trying to do different things.

spase74, 2019-11-14:
There are two versions of nokia5110 available, 5v and 3.3v. Controller itself works at 3.3v, and extra level converters are mounted on supplying PCB in case of 5v device. Check whether this is your case.

closed 2020-01-18 by olikraus