olikraus/u8g2 #1044: ssd1327 not work. any device

bookerhsu4026, 2019-11-10:
I use u8g2 helloword demo code to build SSD1327 oled
中景园高清1.5寸oled 128x128液晶屏ssd1327带灰度全程技术支持.
It's not work.
SPI OR I2C .
I try any device in u8g2 , that include ssd1327.
not work.
I run demo code form provider, that's work.

olikraus, 2019-11-10:
Any links? Datasheet? Which code did you use? What exactly does not work? Picture?

olikraus, 2019-11-10:
Can you send a picture of your setup? Which u8g2 constructor did you use?

bookerhsu4026, 2019-11-10:
I buy that from taobao.com.
It run demo code normally.
[中景园高清1.5寸oled 128x128液晶屏ssd1327带灰度全程技术支持](https://item.taobao.com/item.htm?spm=a1z09.2.0.0.31e72e8d28Pm1w&id=568527842788&_u=a1htbi28a946)

bookerhsu4026, 2019-11-10:
But arduino not work with u8g2 or other.
 I try any configuation whith u8g2.
not work.
The screen looks die.
spi or i2c .

bookerhsu4026, 2019-11-10:
the demo code for stm32 with i2c.
------------------------------------
#include "oled.h"
#include "oledfont.h"

void Delay(unsigned int Data)
{
unsigned int temp;
	for(; Data>0; Data--)
	{
		for(temp=0; temp<0xffF; temp++);
	}
}
void IIC_Start()
{

	OLED_SCLK_Set() ;
	OLED_SDIN_Set();
	OLED_SDIN_Clr();
	OLED_SCLK_Clr();
}

/**********************************************
//IIC Stop
**********************************************/
void IIC_Stop()
{
OLED_SCLK_Set() ;
//	OLED_SCLK_Clr();
	OLED_SDIN_Clr();
	OLED_SDIN_Set();
	
}

void IIC_Wait_Ack()
{

	//GPIOB->CRH &= 0XFFF0FFFF;	//扢离PB12峈奻嶺怀耀宒
	//GPIOB->CRH |= 0x00080000;
//	OLED_SDA = 1;
//	delay_us(1);
	//OLED_SCL = 1;
	//delay_us(50000);
/*	while(1)
	{
		if(!OLED_SDA)				//瓚剿岆瘁諉彶善OLED 茼湘陓瘍
		{
			//GPIOB->CRH &= 0XFFF0FFFF;	//扢离PB12峈籵蚚芢轎怀堤耀宒
			//GPIOB->CRH |= 0x00030000;
			return;
		}
	}
*/
	OLED_SCLK_Set() ;
	OLED_SCLK_Clr();
}
/**********************************************
// IIC Write byte
**********************************************/

void Write_IIC_Byte(unsigned char IIC_Byte)
{
	unsigned char i;
	unsigned char m,da;
	da=IIC_Byte;
	OLED_SCLK_Clr();
	for(i=0;i<8;i++)		
	{
			m=da;
		//	OLED_SCLK_Clr();
[... 335 lines cut ...]
Write_Register(0xa4);//Normal Display
Write_Register(0xa8);//Set multiplex ratio
Write_Register(0x7f);
Write_Register(0xab);//Function Selection A
Write_Register(0x01);//Enable internal VDD regulator
Write_Register(0x81);//Set contrast
Write_Register(0x77);
Write_Register(0xb1);//Set Phase Length
Write_Register(0x31);
Write_Register(0xb3);//Set Front Clock Divider /Oscillator Frequency
Write_Register(0xb1);
Write_Register(0xb4); //For brightness enhancement
Write_Register(0xb5);
Write_Register(0x03);
Write_Register(0xb6);//Set Second pre-charge Period
Write_Register(0x0d);
Write_Register(0xbc);//Set Pre-charge voltage
Write_Register(0x07);
Write_Register(0xbe);//Set VCOMH
Write_Register(0x07);
Write_Register(0xd5);//Function Selection B
Write_Register(0x02);//Enable second pre-charge
Write_Register(0xaf);//Display on

}

olikraus, 2019-11-10:
Which u8g2 constructor did you use?

bookerhsu4026, 2019-11-10:
I try all u8g2 constructor  for ssd1327.
don't work.

olikraus, 2019-11-10:
Usually you need to update the arguments and assign the correct pin numbers.

bookerhsu4026, 2019-11-10:
Yes, I done.

olikraus, 2019-11-10:
ok, nevertheless, I think you should use SW I2C together with the correct pins. 
```
U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, clock, data )
```
for "clock" and "data" you must provide the correct pin number.

bookerhsu4026, 2019-11-11:
demo :

bookerhsu4026, 2019-11-11:
demo code.
Analyzer Name, Decoded Protocol Result
I2C,Setup Write to [0x40] + ACK
I2C,0x0C + ACK
I2C,0x0C + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x36 + ACK
I2C,0x36 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x40 + ACK
I2C,0x40 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x66 + ACK
I2C,0x66 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x10 + ACK
I2C,0x10 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x08 + ACK
I2C,0x08 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x2E + ACK
I2C,0x2E + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x3B + ACK
I2C,0x3B + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x14 + ACK
I2C,0x14 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x0E + ACK
I2C,0x0E + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x17 + ACK
I2C,0x17 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x52 + ACK
I2C,0x52 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x18 + ACK
I2C,0x18 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x10 + ACK
I2C,0x10 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x32 + ACK
I2C,0x32 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x42 + ACK
I2C,0x42 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x1C + ACK
I2C,0x1C + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x10 + ACK
I2C,0x10 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x60 + ACK
I2C,0x60 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x58 + ACK
I2C,0x58 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x20 + ACK
I2C,0x20 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0xED + ACK
I2C,0xED + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x3F + ACK
[... 4695 lines cut ...]
2.994070000000000,I2C,Setup Read to [0x41] + ACK
2.994125000000000,I2C,0x24 + ACK
2.994180000000000,I2C,0x24 + NAK
2.994380000000000,I2C,Setup Read to [0x49] + ACK
2.994435000000000,I2C,0x4F + ACK
2.994490000000000,I2C,0x4F + NAK
2.999565000000000,I2C,Setup Read to [0x49] + ACK
2.999620000000000,I2C,0x71 + ACK
2.999675000000000,I2C,0x71 + NAK
3.001250000000000,I2C,Setup Write to [0x40] + ACK
3.001305000000000,I2C,0x3C + ACK
3.001360000000000,I2C,0x3C + NAK
3.001635000000000,I2C,Setup Read to [0x41] + ACK
3.001690000000000,I2C,0x31 + ACK
3.001745000000000,I2C,0x31 + NAK
3.001945000000000,I2C,Setup Read to [0x49] + ACK
3.002000000000000,I2C,0x7F + ACK
3.002055000000000,I2C,0x7F + NAK
3.007130000000000,I2C,Setup Read to [0x49] + ACK
3.007185000000000,I2C,0x7D + ACK
3.007240000000000,I2C,0x7D + NAK
3.008815000000000,I2C,Setup Write to [0x24] + NAK
3.009090000000000,I2C,Setup Write to [0x40] + ACK
3.009145000000000,I2C,0x40 + ACK
3.009200000000000,I2C,0x40 + NAK

bookerhsu4026, 2019-11-11:
sorry this.

Analyzer Name, Decoded Protocol Result
I2C,Setup Write to [0x40] + ACK
I2C,0x0C + ACK
I2C,0x0C + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x36 + ACK
I2C,0x36 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x40 + ACK
I2C,0x40 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x66 + ACK
I2C,0x66 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x10 + ACK
I2C,0x10 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x08 + ACK
I2C,0x08 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x2E + ACK
I2C,0x2E + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x3B + ACK
I2C,0x3B + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x14 + ACK
I2C,0x14 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x0E + ACK
I2C,0x0E + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x17 + ACK
I2C,0x17 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x52 + ACK
I2C,0x52 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x18 + ACK
I2C,0x18 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x10 + ACK
I2C,0x10 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x32 + ACK
I2C,0x32 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x42 + ACK
I2C,0x42 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x1C + ACK
I2C,0x1C + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0x10 + ACK
I2C,0x10 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x60 + ACK
I2C,0x60 + NAK
I2C,Setup Read to [0x49] + ACK
I2C,0x58 + ACK
I2C,0x58 + NAK
I2C,Setup Write to [0x40] + ACK
I2C,0x20 + ACK
I2C,0x20 + NAK
I2C,Setup Read to [0x41] + ACK
I2C,0xED + ACK
I2C,0xED + NAK
I2C,Setup Read to [0x49] + ACK
[... 4696 lines cut ...]
2.994070000000000,I2C,Setup Read to [0x41] + ACK
2.994125000000000,I2C,0x24 + ACK
2.994180000000000,I2C,0x24 + NAK
2.994380000000000,I2C,Setup Read to [0x49] + ACK
2.994435000000000,I2C,0x4F + ACK
2.994490000000000,I2C,0x4F + NAK
2.999565000000000,I2C,Setup Read to [0x49] + ACK
2.999620000000000,I2C,0x71 + ACK
2.999675000000000,I2C,0x71 + NAK
3.001250000000000,I2C,Setup Write to [0x40] + ACK
3.001305000000000,I2C,0x3C + ACK
3.001360000000000,I2C,0x3C + NAK
3.001635000000000,I2C,Setup Read to [0x41] + ACK
3.001690000000000,I2C,0x31 + ACK
3.001745000000000,I2C,0x31 + NAK
3.001945000000000,I2C,Setup Read to [0x49] + ACK
3.002000000000000,I2C,0x7F + ACK
3.002055000000000,I2C,0x7F + NAK
3.007130000000000,I2C,Setup Read to [0x49] + ACK
3.007185000000000,I2C,0x7D + ACK
3.007240000000000,I2C,0x7D + NAK
3.008815000000000,I2C,Setup Write to [0x24] + NAK
3.009090000000000,I2C,Setup Write to [0x40] + ACK
3.009145000000000,I2C,0x40 + ACK
3.009200000000000,I2C,0x40 + NAK

closed 2019-11-11 by bookerhsu4026

bookerhsu4026, 2019-11-12:
I see that, not standard i2c protocol.

olikraus, 2019-11-13:
I do not know how to support here. :-(

You did not provide information about your setup and you did not provide the way how u8g2 is initalized. So, I can not crosscheck this and so i can not give you any solution.

Sending tons of c code and protocols also does not help. 

Additionally, you have closed this issue, so I assume the problem is solved.

bookerhsu4026, 2019-11-13:
I try Hello World! example. no work.
this
U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 5, /* data=*/ 4, /* reset=*/ U8X8_PIN_NONE);
and this.
U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2019-11-13:
Are the pin numbers 5 and 4 correct? What is your board? How are pins mapped from Arduino to GPIO ports?

bookerhsu4026, 2019-11-15:
What is your board? ESP8266 nodemcu
D1 -? clock
D2 -> data
D4 -> reset
I try another pins. not work too.

olikraus, 2019-11-15:
This does not fit together. Above you said:
```
U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=/ 5, / data=/ 4, / reset=/ U8X8_PIN_NONE);
```
and in your last post:

> D1 -? clock
D2 -> data
D4 -> reset

This seems to be wrong....

```
U8G2_SSD1327_EA_W128128_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 1, /* data=*/ 2, /* reset= */ 4);
```