olikraus/u8g2 #1046: No display with Hardware I2C on SSD1306_128x64

lenceliu, 2019-11-14:
The original SSD1306_128x64 display with 7 Pin and using SPI communication. Since another SPI device conflict with SSD1306, so I move the resistor R3 to R1 which on the SSD1306 board and the SSD1306 communication changed from SPI to I2C.
Unfortunately, I have a INA219 also connect with I2C. The issue is:
1. U8G2_SSD1306_128X64_NONAME_F_SW_I2C is work on display alone, but if ina219.begin() is applied, there is no display.

2. U8G2_SSD1306_128X64_NONAME_F_HW_I2C and U8G2_SSD1306_128X64_ALT0_F_HW_I2C not work on display whether applied ina219.begin() or not.

3. The U8glib is worked if I use U8GLIB_SSD1306_128X64 u8g(U8G_I2C_OPT_NO_ACK), but too short amount of fonts on U8glib, so I still want to use U8g2lib.

olikraus, 2019-11-14:
You must use HW_I2C, because SW_I2C will conflict with other devices using hardware based I2C: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L9

U8g2 will increase the bus clock. Maybe you should set the busclock to 100000Hz only: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

lenceliu, 2019-11-14:
I am very appreciate your early reply.

As your said, I definitely need to use HW_I2C in this project.

I tried u8g2.setBusClock(100000) in "HelloWorld" example with the definition of U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE)
The display still not work.

Is this issue related to the display not a original I2C board. As previous mentioned, I changed the resistor R3 to R1 to get it from SPI to I2C.

olikraus, 2019-11-28:
Oh, I just saw your remark on NO_ACK: your display is broken. It will work only in SW i2c mode.

closed 2020-01-18 by olikraus