olikraus/u8g2 #1047: Request for SSD1306 72x40 oled support

nikthefix, 2019-11-15:
There are new modules all over ebay and aliexpress using this configuration:

https://www.aliexpress.com/item/33047094998.html?spm=a2g0o.productlist.0.0.a0636686fuUxju&algo_pvid=f9be7ce0-c128-4a1e-85d5-58f4653bd0f6&algo_expid=f9be7ce0-c128-4a1e-85d5-58f4653bd0f6-0&btsid=77a37fc2-c177-48fe-8448-2333f335f89d&ws_ab_test=searchweb0_0,searchweb201602_9,searchweb201603_52

I've modified the SH1106 code to support this in my code but it would great to have you include it properly in the lib.

Thanks for the great work!

olikraus, 2019-11-15:
Any reason why you have modified the SH1106 code? Shouldn't it be SSD1306 code instead?
Can you share your modifications?

olikraus, 2019-11-15:
I have ordered one of these OLEDs

nikthefix, 2019-11-16:
Hi,

Yes I actually did both just to see which would be the most sensible
starting point and ended up using ssd1306 64x32 with some override commands
in the main sketch (just raw wire.writes) so that I didn't need to change
the libs at all. With this display module there is one BIG issue. There is
no external IREF so using any of the popular SSD1306 libs will produce a
very dim display. During the init phase we need to send:

Command 0xAD //IREF mode
Command 0x30// 0=External ...or  b00110000=Internal where A[4] sets
internal mode and A[5] sets the current reference to 19uA or 30uA giving
150uA and 240uA segment currents respectively. The display is really bright
at 240uA with a good diming range from the contrast control.

None of the libraries I've looked at cater for this - maybe because this
feature is only documented in SSD1306B v1.1 data sheet and not in SSD1306
v1.

Another oled which will soon be hitting ebay/ ali  etc in module form (I
have a man on the inside!) is the SSD1306 72x32 0.41". I've ordered a few
of the bare screens to mount to a custom pcb and will experiment and
compare. There is also a SSD1331 64x64 colour oled which I'm going to try -
not found a mounted version of this yet so may have to build this too.

I'm a huge fan of your work.

Many Thanks

Nick

nikthefix, 2019-11-16:
BTW, this is the module I was referring to and suspect it's the one you've
ordered:

https://www.aliexpress.com/i/33047094998.html?ws_ab_test=searchweb0_0,searchweb201602_0_10130_453_454_10618_536_317_537_319_10059_10696_10084_10083_10547_10546_10887_10307_321_10548_322_10065_10068_10103_10884_10545,searchweb201603_0,ppcSwitch_0&algo_pvid=6ce2eb17-0771-4255-bd4c-c8cb8d541c57&algo_expid=6ce2eb17-0771-4255-bd4c-c8cb8d541c57-4

Other variants of this module may have external IREF as is the case with
the other popular SSD1306 oled modules. I guess using internal IREF reduces
component count a little and allows user control of max brightness in
software.

Nick

olikraus, 2019-11-16:
I have ordered this display: https://www.ebay.com/itm/I2C-White-0-42-inch-OLED-Display-Module-72x40-Arduino-Raspberry-Pi-W-SSD1306/293291415905

nikthefix, 2019-11-17:
Yes this is what i have except that yours shows an early iteration misprint
in the silk screen - 70x40 instead of 72x40.

Nick

olikraus, 2019-11-17:
Oh, you mean it is actually 72x40? Let's see then...

epicboss200, 2019-12-03:
@olikraus @nikthefix Bump... Just got an SSD1306 0.42 72x40 display, initially tried with the Adafruit SSD1306 library, but then found a custom library from an OLED manufacturer in China that works. Really want to use U8G2 on the display though!

Any plans on adding support for this display soon?

epicboss200, 2019-12-03:
> Hi, Yes I actually did both just to see which would be the most sensible starting point and ended up using ssd1306 64x32 with some override commands in the main sketch (just raw wire.writes) so that [...]

Hi @nikthefix , greetings from India!

Just asking for some friendly advice. What's the best way to proceed with an SSD1306 72X40 display? I've assembled it onto a custom PCB. We're building a smart glass for people with disabilities! 

Also, I did notice that the display is quite dim compared to other displays. Could you please elaborate a bit on how to fix that (I'm not extremely proficient at Arduino/libraries/hardware displays, but might be able to figure this out with some guidance from you)

Thanks so much!
Madhav

olikraus, 2019-12-03:
The display didn't reach my lab until now. It usually takes more than 20 days for parts from far East to reach Europe.

olikraus, 2019-12-03:
You won't believe, I just checked my mail (for sure I check it every day), so this little thing reached me today morning:
[image]

As usualy buydisplay added a nice huge paper box around it, probably 20 times to large, but obviously perfectly protected for a long journey.

nikthefix, 2019-12-03:
Hi Madhav,

All you need to do to get the higher brightness  is add two commands to the display init routine in whichever library you are using.

0xAD //iref
0x30 //set to internal ref @240uA

You can add these in the library at the location where the multiplex ratio is set.

See the attached examples which I can confirm work correctly with this display (both spi and i2c versions provided).
[ER-OLEDM0.42-1_Arduino_Tutorial.zip](https://github.com/olikraus/u8g2/files/3917804/ER-OLEDM0.42-1_Arduino_Tutorial.zip)

epicboss200, 2019-12-03:
Haha what an absolute coincidence!!

Can't believe it.

Look forward to you testing it out :))

Hope to hear from you on updates soon.

Best,
Madhav

olikraus, 2019-12-03:
All the constructors are there, but i will add the HW_I2C constructors to the examples only...
```
U8G2_SSD1306_72X40_ER_F_HW_I2C
U8G2_SSD1306_72X40_ER_1_HW_I2C
U8X8_SSD1306_72X40_ER_HW_I2C 
```

epicboss200, 2019-12-03:
Is it possible to add SPI support?

Hardware and software SPI?

Thanks for such quick responses. Really a fan of all your work

v1993, 2019-12-03:
@epicboss200 I might be wrong, but for me it looks like display olikraus got have no SPI support based on available pins.

olikraus, 2019-12-03:
> @epicboss200 I might be wrong, but for me it looks like display olikraus got have no SPI support based on available pins.

True, but the SPI constructors are automatically generated.

olikraus, 2019-12-03:
The SPI constructors are documented here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-72x40_er

epicboss200, 2019-12-03:
Yes, I see Olikraus display doesn't have SPI pins, but there are several
modules/displays with this resolution that do. I have a 0.42 72*40 SSD1306
with SPI.

Are you saying that if I call the constructor for I2C you mentioned above,
the SPI constructors are generated?

olikraus, 2019-12-03:
Gallery entry:
https://github.com/olikraus/u8g2/wiki/gallery#3-dec-2019-ssd1306-72x40-042-oled

> the SPI constructors are generated?

of course... : https://github.com/olikraus/u8g2/commit/00ad8eaa91ca4007a970e51bba097866699ba838

olikraus, 2019-12-03:
Thanks a lot for pointing out this new display. I really like this small but active display community :-)

I have created beta 2.27.3 with support for the above mentioned display (I2C and SPI interfaces!)

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-12-03:
Hmm... I think there is an issue with large wiki pages:

The wiki page took too long to render.
Please edit this wiki page’s content so it renders faster.

https://github.com/olikraus/u8g2/wiki/u8g2setupcpp will not load at the moment.

epicboss200, 2019-12-03:
Great work, Olikraus!

I'm going to have some fun with this library on these new displays, will
let you know how it goes!

Honestly, OLED displays are super interesting to me because of the vast
possibilities. I think Nick has some good inside information on new oleds :)

epicboss200, 2019-12-03:
Hi Nick,

Thanks so much for the help!

Thanks also for the library and example :)

I'm going to explore this and let you know if I have any questions/issues.

Thanks,
Madhav

epicboss200, 2019-12-04:
> Hi Madhav,
>
> All you need to do to get the higher brightness is add two commands to the display init routine in whichever library you are using.
>
> 0xAD //iref
> 0x30 //set to internal ref @240uA
>
> You [...]

Hi @nikthefix 

I had already tried this library (found it from ERs display product page on BuyDisplay). I can see in the library cpp file that these hex commands are already there. However, my display has the same brightness with this library/initialization commands.. There's no difference.

I am running the OLED soldered onto on a custom PCB without an external IREF.

Are you sure these commands are correct? I saw the command table in the datasheet, and found external/internal IREF setting hex AD, but couldn't find commands to set A[5] such that Iseg is 240uA.

Can you help me out here?

Are the commands correct for setting IREF internal mode and setting IREF such that ISEG is set to 240uA max?

> Command 0xAD //IREF mode
Command 0x30// 0=External ...or  b00110000=Internal where A[4] sets
internal mode and A[5] sets the current reference to 19uA or 30uA giving
150uA and 240uA segment currents respectively. The display is really bright
at 240uA with a good diming range from the contrast control.

nikthefix, 2019-12-04:
Bit 5 of the 2nd byte sets the internal current reference value so:

0x30 = internal / 240uA (b00110000)
0x10= internal / 150uA (b00010000)

Have you tried changing the Vcom Deselect level (absolutely not necessary
with the above mentioned module but a useful experiment perhaps)

0xDB
0x20 - power on default - 77% of vcc

try

0xDB
0x30 - 83% of vcc

Changing either of these settings should make a clear difference. If not
then perhaps you could upload a schematic of your carrier board.
Are unused pins tied according to the datasheet where indicated?
How does your board differ from the ER-OLEDM0 if at all?
Decoupling of power?
Are you powering 3.3v or 5v thru a regulator?
If using SPI is there anything else sharing the bus?

I have this display (both the module and bare display soldered to FFC
adapter and breadboarded) working as brightly if not brighter than my other
128x64 ssd1306 displays using these int codes. The data sheet stipulates
that these settings should be transmitted with the display in sleep mode
(Display Off Command - Init codes - Display On Command).

I can't think of any other likely culprits but please report back with your
findings.

Nick

nikthefix, 2019-12-04:
BTW Madhav,

Have you checked this display out?

https://www.aliexpress.com/item/4000245224527.html?spm=a2g0o.productlist.0.0.3c843964tCSVGS&algo_pvid=d8615d5c-d08a-4176-910c-ed32b1cdb97e&algo_expid=d8615d5c-d08a-4176-910c-ed32b1cdb97e-17&btsid=1757c05f-2128-423b-b6be-5e9e988dc216&ws_ab_test=searchweb0_0,searchweb201602_5,searchweb201603_52

1024×768 Colour 0.5" !!!

Expensive but for your HUD / smart glasses could make a good MK2.

closed 2020-01-18 by olikraus