olikraus/u8g2 #1050: Request for Winstar WEA009664B oled support (SSD1327 96x64 I2C)

baglsim, 2019-11-19:
Hi, 

Unfortunately I didn't found any correct HW configuration for the Winstar OLED SSD1327 96x64 (I2C interface). Here the link: https://www.winstar.com.tw/it/products/oled-module/graphic-oled-display/96x64-oled.html

I'm alredy using, without any problems, the Winstar WEO006448A (SSD1306 64x48 I2C). 

Thanks for great job!

Regards,
Simone

olikraus, 2019-11-19:
Can you test one of the existing SSD1327 constructors with the 96x64 display? 
Which constructor will work best? I actually need a starting point for the new constructor.

baglsim, 2019-11-19:
Hi,
thanks for your fast answer.

With u8g2_Setup_ssd1327_i2c_seeed_96x96_f some commands are working (eg. u8g2_ClearDisplay) ...but the function u8g2_DrawStr doesn't work at all.

Following the constructor code with the correct initialization in SPI mode: [WEO009664B-SPI.zip](https://github.com/olikraus/u8g2/files/3864603/WEO009664B-SPI.zip).

I tried to adapt it with STM32 and HAL library in I2C mode and works perfectly. But I'd like to use your library instead.

Best Regards,
Simone

olikraus, 2019-11-19:
Can you attach a picture of the 96x96 constructor?
Maybe you can also attach the modified code.

baglsim, 2019-11-19:
Here the modified code, it works well with stm32f4 and I2C HAL library (polling mode). 
[winstar_ssd1327_96x48.zip](https://github.com/olikraus/u8g2/files/3865285/winstar_ssd1327_96x48.zip)

Here the picture with seeed 96x96 constructor after the u8g2_ClearDisplay (with 0xff instead of 0x00).
[image]

Regards,
Simone

baglsim, 2019-11-19:
Here the source code with the right OLED size in the name (96x64):
[winstar_ssd1327_96x64.zip](https://github.com/olikraus/u8g2/files/3865493/winstar_ssd1327_96x64.zip)

BR
Simone

olikraus, 2019-11-23:
I have added "ssd1327_ws_96x64" constructor functions

```
void u8g2_Setup_ssd1327_ws_96x64_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
void u8g2_Setup_ssd1327_i2c_ws_96x64_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
```

Can you test this?

baglsim, 2019-11-25:
Hi,
I just tested the new code (use U8G2_R2 to see the right display view):

 u8g2_t u8g2; // a structure which will contain all the data for one display

  u8g2_Setup_ssd1327_i2c_ws_96x64_f(&u8g2, U8G2_R2, u8x8_byte_stm32hal_hw_i2c, psoc_gpio_and_delay_cb);  // init u8g2 structure

  u8g2_InitDisplay(&u8g2);
  u8g2_SetPowerSave(&u8g2, 0);
  u8g2_ClearDisplay(&u8g2);

  u8g2_SetFont(&u8g2, u8g_font_5x8);
  u8g2_DrawStr(&u8g2, 10,10,"Hello World!");
  u8g2_SendBuffer(&u8g2);

The buffer fill the screen but there is a side not correctly initialized (picture follwing). 

[image]

BR
Simone

baglsim, 2019-11-25:
I found:

/* default_x_offset = */ 16, // instead of 0
/* flipmode_x_offset = */ 16, // instead of 0

And a little mistake in initialization (before was 63 decimal):
U8X8_CA(0x0a8, 0x063),		/* multiplex ratio: 63* 1/64 duty */

After these correction all seems to run fine (picture below - u8g2_font_pressstart2p_8f)!
[image]

Now I will take some times to test everything, but I can consider closed the Issue.
Thanks again for your great job!

BR
Simone

olikraus, 2019-11-25:
Thanks for your detailed feedback. 

I added the changes to the repo (marked as green):
https://github.com/olikraus/u8g2/commit/beed2a2c4a0ef3c04d0afb6a52a0c267b4fedc77
Are these changes correct?

baglsim, 2019-11-25:
Hi,
Just updated the rep, the changes are working perfectly.
Thanks!

BR
Simone

olikraus, 2019-11-25:
great, thanks for the quick feedback.

baglsim, 2019-11-25:
Just another "quick" info, what I should take care to "optimize" my display refresh?

I mean, I would like update my display every 1s (it takes about 110ms via I2C), but only when the buffer changes. Exist some simple mechanism to call u8g2_SendBuffer() only when something changes?

Moreover...if something changes, it is possible update only the "buffer part" that changes?

Obviously I would like to save some time :).

Example below:

> char disp_buffer[20];
> extern float cpu_temp;
>  while  (1)
>  {
>     sprintf(disp_buffer, "Cpu Temp: %0.2f ", cpu_temp);
>     u8g2_DrawStr(&u8g2, 0,10, disp_buffer);
>     **if (__is_internal_buffer_different__)**
>       u8g2_SendBuffer(&u8g2); // 
>   }

Thanks in advance.

BR
Simone

olikraus, 2019-11-26:
There is no such internal flag. Instead I suggest to store the old value of "cpu_tmp" and if that value is different, then only execute the draw string and the send buffer command.

baglsim, 2019-11-26:
Hi,
my real case is less "easy" then previous example, but anyway I can find another solution.

Only for info I have in mind something similar:

> static char TempBuffer[768] = {0}; // static screen size
> 
> if (memcmp(TempBuffer, u8g2_GetBufferPtr(&u8g2), sizeof(TempBuffer)))
> {
>   u8g2_SendBuffer(&u8g2);
>   memcpy(TempBuffer, u8g2_GetBufferPtr(&u8g2), sizeof(TempBuffer));
> }
> 

BR
Simone

closed 2019-11-26 by baglsim

olikraus, 2019-11-26:
nice approach...