olikraus/u8g2 #1051: C font structure?

e-labInnovations, 2019-11-19:
@olikraus 
Please give me a documentation about U8g2 c file font structure

olikraus, 2019-11-19:
1: https://github.com/olikraus/u8g2/wiki/u8g2fontformat

2: There are two approaches
2.1: Via bdfconv and
2.2: Via the TGA device: https://github.com/olikraus/u8g2/tree/master/sys/tga

The generation itself is done here:
https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c#L1989

e-labInnovations, 2019-11-19:
@olikraus 
Thank you

e-labInnovations, 2019-11-19:
@olikraus  

## Can I decode it ?
## convert into bdf

`/*
  Fontname: -Misc-Fixed-Bold-R-Normal--18-120-100-100-C-90-ISO10646-1
  Copyright: Public domain font.  Share and enjoy.
  Glyphs: 18/762
  BBX Build Mode: 2
*/
const uint8_t u8g2_font_9x18B_mn[309] U8G2_FONT_SECTION("u8g2_font_9x18B_mn") = 
  "\22\2\4\3\4\5\1\2\5\11\20\0\376\16\374\16\374\0\0\0\0\1\30 \10\11\223\37\377\22\0*"
  "\21\11\223\37c!\11%FD&\22\212\344Q\0+\16\11\223\37c\71E\7Ar\352Q\0,\12"
  "\11\223\37\77['&\4-\11\11\223\37\357\356\61\5.\11\11\223\37\77[\17\6/\20\11\223\37\261"
  "\230\234\230:\61ub\362\240\0\60\30\11\223\37\345\224\204\214\22\31\21\31\21\31\21\31\21\31\25R\363"
  "`\0\61\20\11\223\37\245\330\24\221\204\234\36\35\304\3\1\62\17\11\223\37\345\224\204\214:e:\273\7"
  "\2\63\20\11\223\37\335\235\230\262Au\42\212\350\301\0\64\24\11\223\37\255\330\24\221\204\214\210\214\210\314"
  "\231\234z(\0\65\20\11\223\37\335\211\234\272Bu\42\212\350\301\0\66\24\11\223\37%\221\230\234\134\221"
  "\210\214\22\31\25R\363`\0\67\20\11\223\37\335\235\230\234\230\234\230\234\366p\0\70\25\11\223\37\345\224"
  "\204\214\26RS\22\62JdTH\315\203\1\71\23\11\223\37\345\224\204\214\22\31\25Cu\312\204\350\301"
  "\0:\14\11\223\37g\353\321\214\315\203\1\0\0\0\4\377\377\0";`

olikraus, 2019-11-20:
> Can I decode it ?

probably yes.

> convert into bdf

Maybe, but the BDF is already available, so why converting this back?

e-labInnovations, 2019-11-21:
Thank you @olikraus 
check issue 1053

closed 2020-01-18 by olikraus