olikraus/u8g2 #1052: dynamic allocation u8g2.getBufferSize() results in 0

jaghatei, 2019-11-19:
Hi

I am using dynamic allocation as described in here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#using-dynamically-allocated-page-buffers

But u8g2.getBufferSize() results in 0 so further setup routine is dying in accessing an invalid pointer.

If I replace cpp style u8g2.getBufferSize() with c style u8g2_GetBufferSize(&u8g2) it works.
So error must be anywhere in cpp routine.

I verified the issue with a github snapshot from Nov 9th and from today.

real sketch lines to reproduce:

U8G2_SH1106_128X64_NONAME_F_HW_I2C dsp1(U8G2_R3, U8X8_PIN_NONE);

uint8_t *dsp1buf;

void setup(){ // working setup
  Serial.println(u8g2_GetBufferSize(dsp1.getU8g2())); // results in "1024" on serial
  dsp1buf = (uint8_t*)malloc(u8g2_GetBufferSize(dsp1.getU8g2()));
  dsp1.setBufferPtr(dsp1buf);
  dsp1.initDisplay();
  dsp1.clearDisplay();
  dsp1.setPowerSave(0);
  dsp1.setFont(u8g2_font_missingplanet_tf);
}

void setup(){ // dying setup
  Serial.println(dsp1.getBufferSize()); // results in "0" on serial
  dsp1buf = (uint8_t*)malloc(dsp1.getBufferSize());
  dsp1.setBufferPtr(dsp1buf);
  dsp1.initDisplay();
  // dying here with exception and reboot on ESP
  dsp1.clearDisplay();
  dsp1.setPowerSave(0);
  dsp1.setFont(u8g2_font_missingplanet_tf);
}

olikraus, 2019-11-19:
I do not remember much here. If there is any code inside u8g2 regarding dynamic memory allocation, then it is more or less contributed code.

jaghatei, 2019-11-19:
fix is easier than expected:
Definition of getBufferSize misses "return" in definition:
in U8g2lib.h change
`    uint16_t getBufferSize() { u8g2_GetBufferSize(&u8g2); }`
to
`    uint16_t getBufferSize() { return u8g2_GetBufferSize(&u8g2); }`

olikraus, 2019-11-19:
Oh, this Ivan fix I guess. Thanks for the bugreport.

olikraus, 2020-01-18:
Thanks for the fix.

closed 2020-01-18 by olikraus