olikraus/u8g2 #1053: BDF2U8g2 with JavaScript

e-labInnovations, 2019-11-20:
Hi sir
I am interested to develop a JavaScript module for Nodejs for converting bdf to U8g2 font format. I can get the following details from a bdf font

**Meta data:**
An object containing metadata about the font once loaded. This includes the font version, name, size and several other properties.

Example meta object:

`{
version: '2.1',
  name: 'c64',
  size: { points: 8, resolutionX: 75, resolutionY: 75 },
  boundingBox: { width: 8, height: 8, x: 0, y: -2 },
  properties: { fontDescent: 2, fontAscent: 6, defaultChar: 0 },
  totalChars: 95,
copyright: "free for all"
}`

**glyphs**
An object containing data for every glyph in the font. Each key in this object represents the character encoding.

Example glyphs object:
`{
  ...
  '64': { ... }
  '65': {
    name: 'C0001',
    bytes: [Object],
    bitmap: [Object],
    code: 65,
    char: 'A',
    scalableWidthX: 666,
    scalableWidthY: 0,
    deviceWidthX: 8,
    deviceWidthY: 0,
    boundingBox: { x: 0, y: -2, width: 8, height: 8 }
  }
  '66': { ... }
  ...
}`

The bitmap object corresponding to each glyph contains a matrix of 1s and 0s defining the shape of the glyph in the bounding box. Example bitmap object, for the character 'A' with code 65: (the spaces are 0s, left out in the example below to make it clearer)

`bitmap: [
    [                                                   0, 0, 0,
      1, 1, 0,                                          0, 0
    ],                                                [
      0, 0, 1,                                          1, 1, 1,                                          0, 0                                            ],
    [                                                   0, 1, 1,
      0, 0, 1,                                          1, 0
    ],                                                [
      0, 1, 1,                                          1, 1, 1,
      1, 0                                            ],
    [                                                   0, 1, 1,
      0, 0, 1,                                          1, 0
    ],                                                [
      0, 1, 1,                                          0, 0, 1,
      1, 0                                            ],                                                [                                                   0, 1, 1,
      0, 0, 1,                                          1, 0
    ],                                                [
      0, 0, 0,                                          0, 0, 0,
      0, 0                                            ]
  ],`

The bytes object corresponding to each glyph is similar to the bitmap object, but each series of eight 1s and 0s on a row is encoded in a byte, instead of being laid out as 1s and 0s in an array. Example bytes object, for the character 'A' with code `65: [ 24, 60, 102, 126, 102, 102, 102, 0 ]`

Can I create a U8g2 font with these details?

How can I make?

AnHardt, 2019-11-20:
https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv

There should be all in you need - just in C.

e-labInnovations, 2019-11-21:
Sorry @AnHardt
I am interested to develop a npm module for Nodejs and also interested to develop a online tool for creating U8g2 font from bdf.

olikraus, 2019-11-21:
I assume @AnHardt just wanted to point out, that all required information is there in the C code. This is definitly true. Indeed my time is limited to explain every bit in the code. Best is to analyse the code itself.

olikraus, 2019-11-21:
As mentioned in #1051, there is also a textual description of the format: https://github.com/olikraus/u8g2/wiki/u8g2fontformat

e-labInnovations, 2019-11-21:
Thank you @olikraus

closed 2020-01-18 by olikraus