olikraus/u8g2 #1054: u8x8 Draw_nxN_String fails with wide fonts

SimonJohnCastle, 2019-11-20:
The Draw 1x2 and 2x2 String methods don't take into account the width of the font when using x2, x3 fonts etc. The next x offset is wrong resulting in the second half of the character being overwritten by the next character.

Here are possible fixes.
I tested both issues with two scaled fonts u8x8_font_courB24_3x4_f and u8x8_font_px437wyse700a_2x2_r

static uint8_t u8x8_draw_1x2_string(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s)
{
  uint16_t e;
  uint8_t cnt = 0;
  **uint8_t th = u8x8_pgm_read(u8x8->font+2);**             /* new 2019 format */
  u8x8_utf8_init(u8x8);
  for(;;)
  {
    e = u8x8->next_cb(u8x8, (uint8_t)*s);
    if ( e == 0x0ffff )
      break;
    s++;
    if ( e != 0x0fffe )
    {
      u8x8_Draw1x2Glyph(u8x8, x, y, e);
      **x+=th;**
      cnt++;
    }
  }
  return cnt;
}

static uint8_t u8x8_draw_2x2_string(u8x8_t *u8x8, uint8_t x, uint8_t y, const char *s)
{
  uint16_t e;
  uint8_t cnt = 0;
  **uint8_t th = u8x8_pgm_read(u8x8->font+2);**             /* new 2019 format */
  **uint8_t xs = 2 * th;** // calc once
  u8x8_utf8_init(u8x8);
  for(;;)
  {
    e = u8x8->next_cb(u8x8, (uint8_t)*s);
    if ( e == 0x0ffff )
      break;
    s++;
    if ( e != 0x0fffe )
    {
      u8x8_Draw2x2Glyph(u8x8, x, y, e);
      **x+=xs;**
      cnt++;
    }
  }
  return cnt;
}

olikraus, 2019-11-20:
nice finding...

olikraus, 2020-01-19:
hmm... i think this can be closed...

closed 2020-01-19 by olikraus