olikraus/u8g2 #1055: Request for u8x8.drawTileP()

SimonJohnCastle, 2019-11-20:
It would be nice to have a version of drawTile() that can take a PROGMEN pointer.

So we can use 
const uint8_t img[] PROGMEM = { 0x00, 0x00, 0x00, 0x80, 0x80 ... ...
we could also use F("\x12\x23\x56\x00\x12....  ....")

Thank you.

... Simon

olikraus, 2019-11-20:
Wouldn't a local procedure in your ino script solve the problem?

```
void drawTileP(uint8_t x, uint8_t y, uint8_t cnt, uint8_t *tile_ptr_p) 
{
  uint8_t buf[8];
  memcpy_P(buf, tile_ptr_p);
  u8x8.drawTile(x,  y,  cnt, buf);
} 
```

SimonJohnCastle, 2019-11-20:
sure I'm doing something like that, I have 4 lines of 3chars, so the buffers a bit bigger but it works fine.
I then have some groups of tiles that I overlay for animation.
Just thought It may be nice to have it in the main library for all, and avoid the second copy.
I did look into your code, so realize it would require updating quite a few drivers to get it in so I can imagine you may not do it if there is an easy work around as you suggest.
Thanks for lib

mianqi2016, 2019-11-22:
> 
> 
> Wouldn't a local procedure in your ino script solve the problem?
> 
> ```
> void drawTileP(uint8_t x, uint8_t y, uint8_t cnt, uint8_t *tile_ptr_p) 
> {
>   uint8_t buf[cnt];  // not 100% whether this dynamic array alloc is ok...
>   memcpy_P(buf, tile_ptr_p, cnt);
>   u8x8.drawTile(x,  y,  cnt, buf);
> } 
> ```

I used your u8g2 library, I want to know what is the difference between SW_SPI and HW_SPI and F_SPI ?

mianqi2016, 2019-11-22:
I'am sorry for ask here, since I couldn't find the contact to you.

olikraus, 2019-11-22:
@mianqi2016 : You can always create a new issue, which I now did for you.

closed 2020-01-18 by olikraus