olikraus/u8g2 #1057: what is the difference between SW_SPI and HW_SPI and F_SPI ?

olikraus, 2019-11-22:
> 
> 
> Wouldn't a local procedure in your ino script solve the problem?
> 
> ```
> void drawTileP(uint8_t x, uint8_t y, uint8_t cnt, uint8_t *tile_ptr_p) 
> {
>   uint8_t buf[cnt];  // not 100% whether this dynamic array alloc is ok...
>   memcpy_P(buf, tile_ptr_p, cnt);
>   u8x8.drawTile(x,  y,  cnt, buf);
> } 
> ```

I used your u8g2 library, I want to know what is the difference between SW_SPI and HW_SPI and F_SPI ?

_Originally posted by @mianqi2016 in https://github.com/olikraus/u8g2/issues/1055#issuecomment-557509879_

olikraus, 2019-11-22:
SW SPI: U8g2 emulates SPI protocol with digitalWrite commands. It will be very slow but also very portable

HW SPI: U8g2 will use SPI.h: This means U8g2 will use the standard SPI functionality.

closed 2020-01-18 by olikraus

mianqi2016, 2020-01-28:
Thanks, in our test, we found u8g is much less than u8g2, our animation program fro LCD12864, could run u8g, and u8g2 was not for lack of FLASH.

Closed #1057<https://github.com/olikraus/u8g2/issues/1057>.