olikraus/u8g2 #1058: undefined reference to `app_main'

lucasexequiel, 2019-11-22:
Hi everyone,
I started a project in Eclipse IDE version 19-03 who works together with ESP-IDF. I want to work with the libraries u8g2 (https://github.com/olikraus/u8g2) and i follow the steps that he propose, in the MAIN FILE of the project building a new file called components and inside open a terminal with the following function "git clone to download the file u8g2" but when i try compile the console show me the following error.

/home/lucas/esp/esp-idf/components/esp32/cpu_start.c:528: undefined reference to `app_main'
collect2: error: ld returned 1 exit status
make: *** [/home/lucas/esp/esp-idf/make/project.mk:483: /home/lucas/esp/workspace/SPI_OK/build/SPI_OK.elf] Error 1
"make all" terminated with exit code 2. Build might be incomplete.

I saw posts and I try adding the next code in my proyect but still not work.

#include <driver/gpio.h>
#include <driver/spi_master.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <stdio.h>
#include "esp_system.h"
#include "nvs_flash.h"
#include <string.h>

#include "u8g2.h" //this is the librarie that i want to add

#include "sdkconfig.h"
//#include "u8g2_esp32_hal.h"

// CLK - GPIO14
#define PIN_CLK 18
// MOSI - GPIO 13
#define PIN_MOSI 19
// CS - GPIO 15
#define PIN_CS 5
// RESET - GPIO 26
#define PIN_RESET 15
// DC - GPIO 27
#define PIN_DC 21

static char tag[] = "test_ST7920";

void task_test_ST7920(void *ignore) {

u8g2_esp32_hal_t u8g2_esp32_hal = U8G2_ESP32_HAL_DEFAULT;
u8g2_esp32_hal.clk   = PIN_CLK;
u8g2_esp32_hal.mosi  = PIN_MOSI;
u8g2_esp32_hal.cs    = PIN_CS;
u8g2_esp32_hal.dc    = PIN_DC;
u8g2_esp32_hal.reset = PIN_RESET;
u8g2_esp32_hal_init(u8g2_esp32_hal);

u8g2_t u8g2; 					// a structure which will contain all the data for one display

u8g2_Setup_st7920_128x64_1(u8g2, rotation, u8x8_byte_8bit_6800mode, uC specific);  // init u8g2 structure

u8g2_InitDisplay(&u8g2); 	 // send init sequence to the display, display is in sleep mode after this,

u8g2_SetPowerSave(&u8g2, 0); 			// wake up display
u8g2_ClearBuffer(&u8g2);
u8g2_DrawBox(&u8g2, 10,20, 20, 30);
u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
u8g2_DrawStr(&u8g2, 0,15,"Hello World!");
u8g2_SendBuffer(&u8g2);

ESP_LOGD(tag, "All done!");

vTaskDelete(NULL);

}

void app_main(){
xTaskCreate(&task_test_ST7920, "task_test_ST7920", configMINIMAL_STACK_SIZE, NULL, 5, NULL)
}

I am trying to handle a ST7920 screen for my proyect and his comunication is in SPI.

olikraus, 2019-11-22:
I personally think, that this is not an u8g2 problem.

v1993, 2019-11-22:
First thins first, please, format your code properly using ```!

Now, this have nothing to do with u8g2. Is this C++ file? Try adding something that should cause an error during compilation to make sure that it is compiled (and hopefully linked as well).

closed 2020-01-18 by olikraus