olikraus/u8g2 #1059: Read bitmap directly from microSD card to display(s) - supported?

limmers2015, 2019-11-22:
I am trying to have my two displays both get their bitmaps from an 8GB microSD card connected via SPI. Here are the constructors I am using, for reference:

`U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R2, /* clock=*/ A1, /* data=*/ A2, /* reset=*/ U8X8_PIN_NONE);`

`U8G2_SSD1327_WS_128X128_1_4W_HW_SPI u8g2b(U8G2_R0, /* cs=*/ 1, /* dc=*/ 0, /* reset=*/ U8X8_PIN_NONE);`

Can I read them directly from the SD card, without having to copy them to SRAM and then to the display? I don't have the SRAM available for a "bitmap buffer" char array like so:

`static const unsigned char Small_20Display_20Graphics_20Prototype_bits[] = {
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xe3, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe1, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xe0, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xf0, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xf8, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0x7e, 0xf6, 0xcf, 0xfe, 0xd9, 0xff, 0xff, 0xff,
   0x3f, 0xf8, 0xff, 0xcf, 0xfe, 0xd9, 0x3f, 0x7b, 0x7e, 0xf2, 0x4f, 0xfe,
   0xc9, 0xff, 0xff, 0xff, 0x1f, 0xfc, 0xff, 0x4f, 0xfe, 0xc9, 0x3f, 0x79,
   0x7e, 0xf2, 0x4f, 0xfe, 0xc9, 0xff, 0xff, 0xff, 0x0f, 0xfe, 0xff, 0x4f,
   0xfe, 0xc9, 0x3f, 0x79, 0x0e, 0xc0, 0x01, 0x38, 0x00, 0xff, 0xff, 0xff,
   0x0f, 0xff, 0xff, 0x01, 0x38, 0x00, 0x07, 0x60, 0x3e, 0xfb, 0x67, 0xff,
   0xec, 0xff, 0xff, 0xff, 0x87, 0xff, 0xff, 0x67, 0xff, 0xec, 0x9f, 0x7d,
   0x3e, 0xf9, 0x27, 0xff, 0xe4, 0xff, 0xff, 0xff, 0x83, 0xff, 0xff, 0x27,
   0xff, 0xe4, 0x9f, 0x7c, 0x3e, 0xf9, 0x27, 0xff, 0xe4, 0xff, 0xff, 0xff,
   0xc1, 0xff, 0xff, 0x27, 0xff, 0xe4, 0x9f, 0x7c, 0xbe, 0xfd, 0xb7, 0xff,
   0xf6, 0xff, 0xff, 0xff, 0xe0, 0xff, 0xff, 0xb7, 0xff, 0xf6, 0xdf, 0x7e,
   0x06, 0xe0, 0x00, 0x1c, 0x80, 0xff, 0xff, 0xff, 0xf0, 0xff, 0xff, 0x00,
   0x1c, 0x80, 0x03, 0x70, 0x9e, 0xfc, 0x93, 0x7f, 0xf2, 0xff, 0xff, 0x7f,
   0xf8, 0xff, 0xff, 0x93, 0x7f, 0xf2, 0x4f, 0x7e, 0x9e, 0xfc, 0x93, 0x7f,
   0xf2, 0xff, 0xff, 0x3f, 0xf8, 0xff, 0xff, 0x93, 0x7f, 0xf2, 0x4f, 0x7e,
   0xde, 0xfe, 0xdb, 0x7f, 0xfb, 0xff, 0xff, 0x1f, 0xfc, 0xff, 0xff, 0xdb,
   0x7f, 0xfb, 0x6f, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x1f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x0f, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x07, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x83,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xc1, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xe1, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xf0,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0x7f, 0xf0, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0x3f, 0xf8, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0x1f, 0xfc,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0x1f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0x0f, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0x07, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0x87, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xc7, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xc0, 0xff, 0xff, 0xff, 0x67, 0x3e, 0xfd, 0xe9, 0x4f,
   0xff, 0x01, 0xc0, 0x81, 0xff, 0xff, 0x3f, 0x86, 0xff, 0xff, 0xff, 0x67,
   0x3e, 0xfd, 0xe9, 0x4f, 0xff, 0xf9, 0xcc, 0x98, 0xff, 0xff, 0x3f, 0xbf,
   0xff, 0xff, 0xff, 0x7f, 0x0e, 0x70, 0x80, 0x03, 0xfc, 0xf9, 0x9c, 0xfc,
   0xff, 0xff, 0x3f, 0xff, 0x03, 0x07, 0x84, 0x67, 0xbe, 0xfe, 0xf5, 0xaf,
   0xff, 0xf9, 0x9c, 0xf8, 0xff, 0xff, 0x3f, 0xfe, 0x33, 0x87, 0x20, 0x67,
   0x9e, 0xfe, 0xf4, 0xa7, 0xff, 0x81, 0xcc, 0xc1, 0xff, 0xff, 0x7f, 0xe0,
   0x79, 0xe6, 0x39, 0x67, 0x06, 0x38, 0xc0, 0x01, 0xfe, 0xf9, 0xc0, 0x8f,
   0xff, 0xff, 0xff, 0x80, 0x79, 0xe6, 0x39, 0x66, 0x5e, 0xfe, 0xf2, 0x97,
   0xff, 0xf9, 0xfc, 0x9f, 0xff, 0xff, 0xff, 0x8f, 0x00, 0xe6, 0x39, 0x67,
   0x4e, 0x7f, 0xfa, 0xd3, 0xff, 0xf9, 0x7c, 0x9a, 0xff, 0xff, 0xff, 0x9f,
   0xf9, 0xe7, 0x39, 0x66, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xf9, 0xfc, 0xc0,
   0xff, 0xff, 0xbf, 0x9f, 0xf9, 0xe7, 0x79, 0x67, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x3f, 0x8c, 0x61, 0xe6, 0x39, 0x66,
   0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x3f, 0xe0,
   0x07, 0xe7, 0x39, 0x67, 0xfe, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f, 0xfe, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x7f,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00 };`

Is this doable? If so, how? I have no idea how to do it.

Thanks in advance for any help!

olikraus, 2019-11-23:
You could takeover and modify the code from DrawXBM https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_bitmap.c#L139

Instead from reading from RAM/ROM you can directly read from the MicroSD card.
Instead of reading via "*b" (https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_bitmap.c#L119) you probably need something like "fgetc()"

limmers2015, 2019-11-23:
I'm gonna be honest here... I'm kinda lost trying to do that. Could you explain how? I know how to edit it, just now what to do. Do I just need to change the references to "*b" to "fgetc(pointer to bitmap file)"? Do I have to change "*bitmap" as well?

Sorry for being so clueless as to what to do... I really appreciate your help! You've been a hero for me so far!

olikraus, 2019-11-23:
Because you had asked so nicely... I have created a complete example: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/LoadFromSD/LoadFromSD.ino

I have created two procedures:
1. writeFile will store a given XBM array to your micro SD card
2. drawFile will read the previously saved file and draw it on the display

I think  the important point is: You have to define your own file format for the file on the SD card. This file format has to be wriiten somehow and later extracted again for the display.

limmers2015, 2019-11-24:
Thanks! This is amazing! I just tested it with my setup and it works perfectly!

Do you have any suggestions as to how to open the files created by this on a computer? That would definitely speed up the process of putting all of my bitmaps in...

Thanks again for all of your help!

olikraus, 2019-11-24:
Probably an external tool is required, which converts s graphics file to that bin format...

olikraus, 2019-11-24:
I have created a small linux commandline tool, which converts a given PNG file to the bin format: https://github.com/olikraus/u8g2/tree/master/tools/png2bin

olikraus, 2019-11-24:
added windows executable

v1993, 2019-11-24:
I think that doing this is pretty bad idea tho. Reading bitmap once and keeping it in memory would be much faster. But if memory is very limited, then this seems ok.

olikraus, 2019-11-24:
No doubt: it's slowly

limmers2015, 2019-11-26:
Thanks a lot for the tool! 

One question: How do I use it? I am seeing that it calls "png2bin.c", and I can see how that works, but how do I point it to the correct file? What are the -l and -o prefixes for?

For reference, I am on macOS, using Xcode or Terminal to run the example.

olikraus, 2019-11-26:
"png2bin.c" is the source code. From that sourcecode I have created a windows executable ("png2bin.exe") and a linux amd64 executable ("png2bin"). Both executables are part of the repo folder: https://github.com/olikraus/u8g2/tree/master/tools/png2bin

I do not have a mac here, so you are on your own to build the software for your OS.

> What are the -l and -o prefixes for?

Can you point out the exact location? Do you mean this?
https://github.com/olikraus/u8g2/blob/beed2a2c4a0ef3c04d0afb6a52a0c267b4fedc77/tools/png2bin/create_png2bdf.sh#L1

If yes, this is used to generate the code under mingw for Windows OS. The options are described as part of the gcc user manual: https://gcc.gnu.org/onlinedocs/gcc/Invoking-GCC.html

closed 2020-01-18 by olikraus