olikraus/u8g2 #1060: how  to display a arduino int variable

lelibreauquotidien, 2019-11-26:
Hello.
All ils a tittle :-).
llaq.

olikraus, 2019-11-26:
And the answer is a link to the manual:

https://github.com/olikraus/u8g2/wiki/u8g2reference#print

lelibreauquotidien, 2019-11-26:
Thanks,and for float variable ?

olikraus, 2019-11-26:
Any variable type will work.

lelibreauquotidien, 2019-11-28:
Yes,it's no work.
No error on compilation but nothing on screen.
llaq.

olikraus, 2019-11-28:
Can you print strings or something else? Do the u8g2 exaples work?

spase74, 2019-11-29:
Try this

`
float f = 1.23f;
char buffer[6];
dtostrf(f, 4 /*total length*/, 2 /*precision*/, buffer);
u8g2.print(buffer);
`
Got from
https://stackoverflow.com/questions/27651012/arduino-sprintf-float-not-formatting

olikraus, 2020-01-18:
u8g2.print() is exactly the same as serial.print(), so float should work...

Nevertheless, closing this issue...

closed 2020-01-18 by olikraus

shakil1740, 2020-10-11:
void loop(void){ 
  int x =12864;
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_osb29_tr);      // set fonts
  u8g2.setCursor(10,45);                           // set cursor position
  u8g2.print(x);	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);
}

shakil1740, 2020-10-11:
void loop(void){ 
float  x =128.64;
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // set fonts
  u8g2.setCursor(30,30);       // set cursor position lcd
  u8g2.print(x,2);	            // write variable to the internal memory
  u8g2.sendBuffer();	    // transfer internal memory to the display
  delay(1000);  
}