olikraus/u8g2 #1061: HW SPI on ESP32 (ST7920)

333fps, 2019-11-26:
Good day,
U8g2lib is working very well on ESP32 WROOM-32 using following constructor:
U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, 14, 12, 13)

I'd like to get it working with Hardware SPI but I cannot find any working constructor.
Am I wasting my time or missing something ?
(Last constructor tested: U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, 13,32); //NOK)

Anyway, thank you for this wonderful library !

v1993, 2019-11-26:
I think that Arduino HAL limit possible HW SPI pins on ESP32.

333fps, 2019-11-26:
> I think that Arduino HAL limit possible HW SPI pins on ESP32.

I didn't care about Arduino before, but maybe that's the point

tobiea27, 2019-11-28:
I'm using an OLED-Display (Driver: SH1106, Res: 128x64) with an ESP32 over HW SPI without any problem.

```
#define OLED_CS             27
#define OLED_DC             26

U8G2_SH1106_128X64_NONAME_F_4W_HW_SPI display(U8G2_R0, /* cs=*/ OLED_CS, /* dc=*/ OLED_DC, /* reset=*/ U8X8_PIN_NONE);//Display SH1106
```

Maybe double check our wiring?
Are you using VSPI? The ESP32 has multiple SPI interfaces and the Pinouts of the Dev-boards can vary .

annhan, 2019-11-29:
Using LSPI with pin 23-mosi and 18-sck. Working for me
```
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0,/*CS=*/ U8X8_PIN_NONE,/*CS=*/U8X8_PIN_NONE);//     
u8g2.begin();
 u8g2.setBusClock(600000);
```

333fps, 2019-11-29:
Thank you very much to all of you, I'll give it a try today.

olikraus, 2019-11-29:
Small remark: If you have to reduce the bus clock, then it might be important to place the setBusClock command before the begin command (https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock):

```
U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0,/*CS=*/ U8X8_PIN_NONE,/*CS=/ U8X8_PIN_NONE); 
u8g2.setBusClock(600000);
u8g2.begin();
```

333fps, 2019-12-01:
Thank you everybody, but I didn't succeed.
Don't worry, I found this TFT screen in a drawer but I plan to replace it with a OLED model (need to print a lot of chars on one line).

closed 2020-01-18 by olikraus

Stephanowicz, 2022-04-09:
Hi,
just stumbled across this while looking for instructions on how to use HW SPI.
(I got it finally running - but after a while the top of the screen gets garbaged...? )

So I think Your problem was that You were using the **SPI2** pins (HSPI) with the **SPI1** (VSPI) instructor!
Here's the pin reference from [components101](https://components101.com/microcontrollers/esp32-devkitc):

```
VSPI | GPIO23   (MOSI), GPIO19(MISO), GPIO18(CLK) and GPIO5 (CS) | Used   for **SPI-1** communication.
HSPI | GPIO13   (MOSI), GPIO12(MISO), GPIO14(CLK) and GPIO15 (CS) | Used   for **SPI-2** communication.

```

So, for **SP1** the instructor is: `U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, 5, U8X8_PIN_NONE);`

(for **SPI2** I dunno which instructor to use - maybe it's U8G2_ST7920_128X64_F_2ND_HW_SPI ? )

Maybe this will help someone ;)

Cheers,

Stephan

harrison3000, 2023-03-19:
I just found a workaround for the HSPI issue, but it only works using a local u8g2 instance
Just leaving it here for anyone to find
Say you have a task named `display` and you want to use the HSPI, just overwrite the global SPI variable before instantiating the display object!
```cpp
#include <SPI.h>

//[...] other things, setup, loop...

void display(){
  SPI = SPIClass(HSPI); //this overwrites the global variable, essencially making HSPI the default
  U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ 27);

  //[...] the rest of display related code
}
```

olikraus, 2023-03-19:
Nice :-)