olikraus/u8g2 #1062: try to print bitmap ont OLED SSD1305

ThorWallhall, 2019-11-29:
I try to print this.........
`#include <Arduino.h>
#include <U8g2lib.h>
#include <SoftwareSerial.h>
#include <DFPlayer_Mini_Mp3.h>

#include <SPI.h>

U8G2_SSD1305_128X32_ADAFRUIT_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

void u8g2_prepare(void){
  u8g2.setFontRefHeightExtendedText();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
}

//#define SCREEN_WIDTH 128
//#define SCREEN_HEIGHT 32

/**
 * Made with Marlin Bitmap Converter
 * http://marlinfw.org/tools/u8glib/converter.html
 *
 * This bitmap from the file 'Lawgiver Eagle (2).png'
 */

#pragma once
#define QF4B47_BMPWIDTH  128
#define QF4B47_BMPHEIGHT 32
static const unsigned char bitmap_qf4b47[] PROGMEM = {
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000001,B11100000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000011,B11111100,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000110,B00011110,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00011111,B00000000,B00001100,B00111000,B00000000,B11111000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00111111,B11100000,B00111000,B01111100,B00000111,B11111100,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B01111111,B11111100,B01110000,B11000111,B00111111,B11111110,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B11110011,B11111111,B11000000,B00001111,B11111111,B11001111,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B11100110,B01111111,B10011111,B11111111,B11111110,B01100111,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B11100100,B01001110,B00001111,B11111000,B01110010,B00100111,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B11101100,B11001000,B00000000,B00000000,B00010011,B00110111,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B11101000,B10011000,B00100010,B01000100,B00011001,B00010111,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B11111001,B10010001,B10001000,B00010001,B10001001,B10011111,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B01110001,B00110011,B10011100,B00111001,B11001100,B10001110,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B01110011,B00100111,B10011100,B00111001,B11100100,B11001110,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00110010,B01100111,B10011100,B00111001,B11100110,B01001100,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00110110,B01001111,B10011100,B00111001,B11110010,B01101100,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00011100,B11001111,B10011100,B00111001,B11110011,B00111000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00011100,B10011111,B11011100,B00111011,B11111001,B00111000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00001111,B10011111,B11111100,B00111111,B11111001,B11110000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00001111,B11111111,B11111100,B00111111,B11111111,B11110000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000111,B11100111,B11111110,B01111111,B11100111,B11100000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000011,B11100011,B11111111,B11111111,B11000111,B11000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000011,B11110011,B11111111,B11111111,B11001111,B11000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000001,B11110001,B10011111,B11111001,B10001111,B10000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B01100000,B00111111,B11111100,B00000110,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B01111111,B11111110,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B11111110,B01111111,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B11111100,B00111111,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B01110000,B00001110,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,
  B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000,B00000000 
};

void testbild(){
  u8g2.clearBuffer();
  u8g2.drawXBMP(0,0,QF4B47_BMPWIDTH,QF4B47_BMPHEIGHT, bitmap_qf4b47);
  u8g2.sendBuffer();
  delay(2000);
}

void setup() {
  // put your setup code here, to run once:
u8g2.begin();
testbild();
}

void loop() {
  // put your main code here, to run repeatedly:

}`

it should look like this....

[image]

but the OLED show me this....

[image]

so where is my mistake?

Hope for help

olikraus, 2019-11-29:
Invert bit order for each byte.

ThorWallhall, 2019-11-29:
how to do this......
at the moment when i convert the png into the bitmap?

ThorWallhall, 2019-11-29:
Sry...could you show me how?

olikraus, 2019-11-29:
B00000111
becomes
B11100000
Same for all other bytes. Obviously
B00000000
Does not change...

ThorWallhall, 2019-11-29:
AAAAHH....ok.....thanks a lot

ThorWallhall, 2019-11-29:
but why does this happen?

olikraus, 2019-11-29:
I do not know your converterbut with Gimp the result would be ok if exported as xbm.

closed 2020-01-18 by olikraus