olikraus/u8g2 #1066: Support Drive uc1628

witc, 2019-12-05:
Dear,

Will you plan add support for driver UC1628 please?

Thank you for information.

Best Regards Jan.

olikraus, 2019-12-05:
Is there any datasheet?

witc, 2019-12-05:
Hi,

Thank you for response. Datasheet is enclosed.

witc, 2019-12-05:
Datashett is there: [UC1628c_A1.0.pdf](https://github.com/olikraus/u8g2/files/3928233/UC1628c_A1.0.pdf)

olikraus, 2019-12-05:
Ah, ok... Is there also a display datasheet which includes the init sequence for your display

witc, 2019-12-06:
There is Init settings:
[image]

olikraus, 2019-12-06:
Are you willing to do some testing here? I mean, I can implement the device, but I can not test it. Usually it requires some bugfixing. Also upload of some pictures of the display output would be required.

witc, 2019-12-12:
OK, first I buy the display and then I write you here that I can test it :) Thank you very much!!

olikraus, 2019-12-12:
hmm... what is special about this display? If you anyway have to buy it, did you look at displays with a ST75256 ot ST75320 controller, which are already supported? 
I mean, I will try to add support for this display, but there is always a risk compared to an existing implementaiton.

witc, 2020-01-18:
> Hi,

I am waiting for delivery of the display, if there will be problem I will
check  displays with ST75256 as you mentioned. Thank you very much.

olikraus, 2020-01-18:
no problem, thanks for the feedback.

olikraus, 2020-06-14:
just open a new issue, if there are some news here...

olikraus, 2021-04-18:
example code: https://www.topwaydisplay.com/sites/default/files/2020-06/LM160160R.txt
datasheet: https://github.com/olikraus/u8g2/files/3928233/UC1628c_A1.0.pdf
discussion: https://github.com/olikraus/u8g2/discussions/1452

olikraus, 2021-04-18:
Oh, there is a second issue: https://github.com/olikraus/u8g2/issues/1453

So you solved this already, right?

olikraus, 2021-06-26:
continued here: https://github.com/olikraus/u8g2/issues/1453

closed 2021-06-26 by olikraus