olikraus/u8g2 #1067: ESP32 Lolin D32 u8g2 initialiation quirky

gritd, 2019-12-06:
I found that if I compile a program and send it to the Lolin D32 from arduino the 128x32 display I connected via I2C initializes fine, but if I then unpug it and use the Lolin standalone on a battery - no display... also replugging it to a computer as well. I need to upload the sketch again and it works. I have the feeling it might be a delay or something but can't put my finger on it... Initialisation is with 
 U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0,22,21);

olikraus, 2019-12-06:
If there is an issue with the startup, you could add a delay before the u8g2.begin() statement.
Also the reduction of the bus clock might help: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

gritd, 2019-12-06:
Hmm well the delay needs to be about 2.5 seconds to be sure about the display turning on which is pretty long... setbusclock did not really change anything. So 2.5s kind of solves it but I hoped there was something better.

olikraus, 2019-12-06:
I assume the display power is connected to 3.3V Can you measure the board voltage if your lipo is connected? Is it really 3.3V?

According to
https://wiki.wemos.cc/_media/products:d32:sch_d32_v1.0.0.pdf
3.3V is derived from the battery via a PMOS and a LDO. Both parts will have a voltage drop. Probably 0.01V for the PMOS and 0.1V for the LDO. It basically should work with a fully loaded Lipo (which has 3.7V).

Actually the OLED is very sensitive to the input voltage. If it is below 3.3V, then the internal change pump requires longer time to start up.

olikraus, 2019-12-06:
One more idea: Add a 1uF ceramic cap next to the display power supply between 3.3V and GND.
Will this improve something?

olikraus, 2020-01-25:
no further feedback, closing...

closed 2020-01-25 by olikraus