olikraus/u8g2 #1068: ST7920 192x64 SPI

umbertoge, 2019-12-06:
Hi and compliments for driver!
Can you add driver for ST7920 for 192x64?

Very thanks
Umberto

olikraus, 2019-12-06:
The 192x64 is a little  bit unusual. Do you have any datasheet of the display itself, which contains the init code?
Are you able to do the testing if I add support for this device?

umbertoge, 2019-12-09:
Hi, yes,
the 192x64 is same 192x32. The difference is an bit for enable low part of 192x32 and an bit for enable high part of 192x32.
If i use your library and enable E1 or E2 the lcd print on hight or low part with no problem.
The modify for library, if y coordinate is > 32 enable E2 disable E1.
See attached.
Thanks

I can test you modify if necesary! Thanks!

[st7920 192x64 example.zip](https://github.com/olikraus/u8g2/files/3938641/st7920.192x64.example.zip)

olikraus, 2019-12-10:
ohhhh, it looks simple to add support for this display, but  it will be a huge effort to change the constructor interface in order to add the additional cs signal.

umbertoge, 2019-12-12:
So, are you going to modify the driver?
Reply asap.
Thanks

olikraus, 2019-12-12:
As I said: It is not only the driver, which needs to be modified. It is the complete u8g2 build environment.
I can say this: I will not be able to do this in the near future.

olikraus, 2020-01-25:
Unfortunately, I do not have the spare time to implement this :-(

closed 2020-06-14 by olikraus

dpwe, 2023-10-22:
I just bought a couple of these displays without realizing they weren't covered by the usual ST7920 interface.  Did anyone ever work on this, or should I take a look?

olikraus, 2023-10-22:
As mentioned above, you could treat the display as two independent 192x32 displays. The u8g2 clipping function would also simplify the task of wrapper functions to use the complete 192x32 display as one huge display.

dpwe, 2023-10-22:
Thanks for the reply.  What's weird about these devices is that the ST7290s *share* the CS pin, and instead are independently addressed by having separate SCLK (E) pins.  Really ugly from an SPI point of view.

So your suggestion is to have a wrapping layer that spans multiple independent u8g2 interfaces but provides duplicates of all the drawing commands, which are then repeated across all the child displays with appropriate clipping/offsets?  That could be nicely flexible for other scenarios.