olikraus/u8g2 #1069: JLX256160 I2C blank with ESP8266

gshegosh, 2019-12-07:
Thank you for a wonderful library which I've been using without issues for years now with Arduino, ESPs and even Raspberry Pi :)

I cannot get my JLX256160 I2C display to work using ESP-12E module as its controller.

Here's my connection list, according to table from JLX spec doc:
```
LCD PIN  | CONNECTION
      1  | none  
      2  | none  
      3  | none
      4  | none
      5  | 3V3
      6  | GND
      7  | 3V3
      8  | 3V3
      9  | ESP-PIN 13 (RST)
     10  | GND
     11  | GND
     12  | GND
     13  | 3V3
     14  | 3V3
     15  | ESP-PIN 4 (SDA, +2K2 pull up resistor to 3V3)
     16  | ESP-PIN 4 (SDA, +2K2 pull up resistor to 3V3)
     17  | ESP-PIN 4 (SDA, +2K2 pull up resistor to 3V3)
     18  | ESP-PIN 0 (SCK, +2K2 pull up resistor to 3V3)
     19  | 3V3
     20  | 3V3
```
Full code - https://gist.github.com/gshegosh/e45a6d1362aa867a46a3118ae53cf71a

I'm using the `U8G2_ST75256_JLX256160_F_HW_I2C u8g2(U8G2_R0, 13, 0, 4)` constructor. Using a simple I2C scanning code, I can see that u8g2 properly initiates Wire and the display can be found at 0x3C address. I have already experimented with:
- setting different levels of contrast and not setting it at all
- explicitly using setI2CAddress(0x3c * 2)
- using software I2C (SW constructor instead of HW, minding param order change)
- changing to other ESP pins
- outputting bus speed, pins and address to Serial in u8x8_byte_arduino_hw_i2c::U8X8_MSG_BYTE_INIT (they're correct)
- outputting bytes sent to Serial in u8x8_byte_arduino_hw_i2c::U8X8_MSG_BYTE_SEND (they seem correct)

Is my display broken or is there anything I'm missing/could try?

olikraus, 2019-12-07:
According to the datasheet of the ST75256 controller, the I2C connection is:
D[0]=SCL, D1=SDA_IN, D[3:2]=SDA_OUT, D[5:4]= ---, D[7:6]=SA[1:0]. 
This means pins 13 and 14  should be unconnected. Not sure whether this makes a difference.

Another question is this: Is the display configure for I2C. And: Is the display configure for 3.3V?
There are some solder bridges to configure the interface. Is it I2C?

[image]

gshegosh, 2019-12-07:
Thank you for your prompt reply. The datasheet for ST75256 I have tells me to wire D[5:4] to HIGH, the configuration jumpers are also correct.

In the end, I've managed to get it working -- I've moved SCL from ESP's pin 0 to pin 5 (the default one). Pin 0 is used during ESP boot and probably messes the display somehow. I haven't used ESP8266s for some time and I've forgotten about it.

Thanks one more time and have a good day. Closing the ticket.

closed 2019-12-07 by gshegosh

olikraus, 2019-12-08:
:)

dynek, 2022-03-27:
@gshegosh did you end up putting a current limiting resistor on pin 5 (backlight) ?

Thank you

gshegosh, 2022-03-29:
> @gshegosh did you end up putting a current limiting resistor on pin 5 (backlight) ?
> 
> Thank you

I don't remember too well, it was more than 2 years ago. I think I've had a small resistor in series with a transistor so that I could control backlight brightness using PWM output.

dynek, 2022-03-29:
Thank you @gshegosh. I made it so I can have the backlight brightness adjusted using PWM as well, using a Darlington configuration. Otherwise PNP is drawing current off the uC when LCD was turned off.

Anyways, the datasheet I have says the backlight needs between 2.9V and 3.1V and draws (measured) abt. 50mA if I remember correctly.
It's powered with 3.3V meaning I should add a .... 6Ω resistor? 🤔

gshegosh, 2022-03-29:
I'd say don't worry about shedding the 0,2-0,4 excess Volts. You need 3.3V / 0,050A = 66 Ohms, the closest being 68 Ohms. This will limit the current in the backlight circuit to about 50mA which will be the maximum brightness and you'll be able to dim it using the transistor. I've used NPN on the low side - the circuit is simpler this way. See discussion at https://electronics.stackexchange.com/questions/303148/high-side-or-low-side-led-driver-with-npn-bjt-which-is-better You should have a resistor on transistor's base which will limit the current to negligible levels.

Please note that I'm a self-taught amateur, so I might be wrong :)

olikraus, 2022-03-29:
The resistor also needs to be strong enough. 3v*0.05mA = 150mW, right? So a standard 1/8 watt resistor would break.

olikraus, 2022-03-29:
I assume 6 ohm are correct, because we need to remove the forward voltage of the led. So it is (3.3 - 3.0)/0.05 = 6 Ohm

dynek, 2022-03-30:
Yesterday I tried with a 6Ω resistor and voltage is around 3V at the backlight pin so I think I'm okay like this.
Also I'd guess only 0.3V will be dissipated by the resistor: P=UI => 0.3*0.05 = 0.015 watts.

Thank you guys!