olikraus/u8g2 #1070: How to create a circular progress bar

alexisalexopoulos, 2019-12-13:
i'm currently working on a project to create a custom gauge in my car. I got most of the code working. The information is coming from the OBD and is beeing displayed on a sh1106 OLED. I'm using different graphics like rectangular progress bars, round gauges etc. I just can't figure out how to create a circular progress bar. What is the best approach? I just the example Gauge.ino to create my own gauges, but this doesn't support the "progress bar"

olikraus, 2019-12-13:
Hm... I have also no idea... Specification is also not clear

alexisalexopoulos, 2019-12-13:
sorry, i'll try to be more specific :). i'm using the Gauge.ino example from https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/Gauge/Gauge.ino. This will draw in my code a half circle, and a line (x,y,xp,yp) where the x and y are fixed positions, and the xp and yp are calculated based on the input value. 

I'm trying to extend this function by creating a circular progress bar that "follows" the line. Just like the following image:

[image]

I'm using a sh1106 Oled screen 128x64.

olikraus, 2019-12-14:
Ok, I understand. I assume you can calculate those pixel with some trigonometry math.

alexisalexopoulos, 2019-12-17:
I'm not a star in math, but if figured it out by creating a for loop in the function. I post it here, maybe it can be useful for someone else.

/*Function DrawgaugeDisc
1 = byte X center position
2 = byte Y center position
3 = byte radius of the gauge
4 = byte needle percentage
5 = int v is the input value
6 = int minVal is the minimum value
7 = int maxVal is the maximum value
8 = float start radians
9 = float end radians
*/
void DrawgaugeDisc(byte x, byte y, byte r, byte p, int v, int minVal, int maxVal, float s_rad, float e_rad) {
	int n=(r/100.00)*p; // calculate needle percent lenght
	for (int a = minVal; a <= v; a++) {
		float i=((a-minVal)*(ge_rad-gs_rad)/(maxVal-minVal)+gs_rad);
		int xp = x+(sin(i) * n);
		int yp = y-(cos(i) * n);
    u8g2.drawLine(x,y,xp,yp);
	}
}

This will create a gauge that will fill the circle based on the input value. 

[image]

olikraus, 2020-01-25:
looks good :-)

closed 2020-01-25 by olikraus