olikraus/u8g2 #1072: Reducing SPI-speed for MAX7219

ld21, 2019-12-16:
Hi

Im using an ESP32 and your u8g2lib with Arduino IDE to drive an 32x8 LED Matrix that is controlled by 4 MAX7219. I have issues that single 8x8-matrices fail to display it's content – even though the ones before or after in the daisy-chaing get their information. The failures don't show up consistently however. Now they are present, after a reboot, they are not and so on. Since this problem reminds me of SPI-speed issues I've had before: I'm wondering if it could be a bus speed problem and wanted to ask if there is an easy way to set the SPI-speed with u8g2lib. Thank you very much,

Dani

PS: Most of the setup is soldered – but of course the fast signals in the wires (see image) could mess up the SPI-data. Reducing the speed would reduce the influence of the different wires to the display.

[image]
[image]
[image]

olikraus, 2019-12-16:
You can change the spi bus speed with the setBusClock (https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock) command.

ld21, 2019-12-17:
Thank you! Do you know of any other reason for single MAX7219s in an chain of MAX7219s to not display any dots after a powerup (as can be seen in the pictures above, where the same image was displayed every time a photo was taken)?

olikraus, 2019-12-17:
my knowledge on the MAX7219 is very limited :-(

closed 2020-01-25 by olikraus