olikraus/u8g2 #1074: Registration of many number of image files.

KOGE-99, 2019-12-17:
Hello. I am writing the code for the LCD remote control. The image is incorporated into the program," static const unsigned char abc_bits{} U8X8_PROGMEM "    but it has now exceeded 200. The file size is 110KB. It suddenly stopped working. It works when reducing image files. MPU is ATMEGA2560. The image file is about 0.1-0.5KB, and the total image file is about 50KB. It is assumed that RAM is still used only about 2KB.No error when compiling. Is there a workaround?

olikraus, 2019-12-17:
There is too less information to give any useful answer.

KOGE-99, 2019-12-17:
Arduino v1.8.10
U8G2 v2.26.14
Megacore v2.0.2
IRREMOTE v2.2.3

ROBOTDYN MEGA2560PROMINI
3.3V 8MHz

LCD 
JLX12864G-COB-F36
ST7567_JLX12864_F_4W_HW_SPI u8g2(U8g2_R0,6,8,7)

Programmer Arduino AS ISP

work
USED Sketch 103126byte
USED RAM 1952Byte
current 3mA

+1KB XBMP  not work
USED Sketch 104174byte
USED RAM 1952Byte
current 25mA
The current increases and nothing is displayed on the LCD. I think the MPU is running away.

olikraus, 2019-12-18:
Maybe there is a stack problem. You need to keep some ram for stack operation.

olikraus, 2020-01-25:
no further feedback, closing...

closed 2020-01-25 by olikraus