olikraus/u8g2 #1077: Does u8g2 support p10 led matrix?

marryton007, 2019-12-18:


olikraus, 2019-12-18:
Probably not

marryton007, 2019-12-19:
The follow article is talking about P10 led matrix  and arduino.
[https://www.instructables.com/id/Display-Text-at-P10-LED-Display-Using-Arduino/](https://www.instructables.com/id/Display-Text-at-P10-LED-Display-Using-Arduino/)

The method of connect has a little special, it bases SPI, but it also has other pins need connect. 

So, I want to know how can I do for supporting P10 led matrix with u8g2? Which things I should need to know?

Thanks.

olikraus, 2019-12-19:
There are instructions how to port u8g2 to a new uC, but there are no instructions on how to add a display with a different wiring. 

If you look at this picture here: https://github.com/olikraus/u8g2/blob/master/doc/u8g2_software_architecture.png

You need to rewrite everything below the U8X8 API (so to say 80% of the picture)

Technically, you need to write your own device procedure. You can find many of these device procedures in the u8g2 src folder, they all start with "u8x8_d_".

closed 2020-01-25 by olikraus