olikraus/u8g2 #1079: Unicode points are 21bits, not 16bits

JeffCalwood, 2020-01-01:
I believe the libraries are using uint16_t for referencing glyphs, but Unicode codepoints really are 21bits. If one is receiving a string (for example UTF-8 encoded) from the internet or via BLE from a smartphone, today one can expect to find Emoji's in there so the library should use a larger datatype.
Thanks.

(P.S. I am using the 'u8g2 fonts for Adafruit_GFX' repo)

olikraus, 2020-01-01:
True, I mentioned this somewhere, but it should be more clear: u8g2 only supports the base multilingual plane. It is a design decision because 32 bit vars are difficult to handle for 8 bit controllers. 16 bit was a good compromise. 

Actually bdfconv supports full Unicode range, so you can map other planes into bmp.

What is your actual problem?

JeffCalwood, 2020-01-01:
The 'actual problem' is, I am using your lib on a nrf52 SOC with bluetooth, and I can expect Emoji's to show up in the UTF8 encoded strings (and I would also like the opportunity to support at least some of them).
I had one or two ideas regarding this, though:
Idea 1: Introduce a typedef "unicode_t" or something to quickly allow changing the type depending on platform.
Idea 2: Since fonts are known at compile time, with some clever #ifdef's you may even be able to adjust the type (from idea 1) automatically, based on the fonts included. In this case, the type for the glyphs would even revert DOWN to uint8_t if the fonts don't include codepoints above that. 
In either case, the code should not fault on larger codepoints .

olikraus, 2020-01-01:
Introduction of a 32bit type is not an option at the moment: The font format itself only uses 16 bit. I mean, the u8g2 font format does not allow the use of 32bit unicode values. This means #defs and typdefs will not help (because the font can not include 32 bit values).

If we would extend the font format from 16 bit to 32 bit, then a font with 200 chars would occupy 400 additional bytes (1/2 kB). I doubt whether Arduino Uno user will accept this. Many people already complain about the size of u8g2. 

It is actually very easy to map for example emoticons into BMP. I did this already with the unifont font, see the u8g2_font_unifont_t_emoticons (somewhere in the middle of this page: https://github.com/olikraus/u8g2/wiki/fntgrpunifont)

The emoticons from the second unicode plane are just mapped into the ASCII range:
https://github.com/olikraus/u8g2/blob/master/tools/font/build/build.c#L709

So with little effort you can solve this problem and save a lot of flash memory.

I can convert the font for you, if you provide the ttf or bdf font file.

JeffCalwood, 2020-01-01:
OK, I could pre-parse my strings and map all the >0xFFFF codepoints which I want to support into the <0xFFFF space. I would use the "private-use" area E000 to F8FF (thanks Wikipedia) for that. That's a compromise.

Even though, Oliver, I want to share that there are very few 'professional grade' libraries out there that have good script rendering for bitmap fonts. I have not found a single library that supports full Unicode range, modifying characters and right to left languages. Maybe Segger's emWin ? To the point that I was tempted to see how big "FreeType" would be on my target. Anyhow. There's an opportunity here :) Thanks for the suggestion above.

olikraus, 2020-01-01:
> right to left languages.

Yes, I was working a little bit on this, but I assume without help of a native speaker, this becomes also very difficult.

I also know, that there are professional libs out there, but... well, this is open source here.
And: Typical Arduino users do not buy high end 32 bit systems (ok, they do become cheaper recently).

U8g2 was designed and still supports systems with very limited resources. 
In fact U8g2 was just a private project to support my own displays, and I didn't want to pay $2000 just to get emWin for a private project. Additionally the driver support will focus more on "what is avialable on the internet market places". I mean, there is some overlap with the drivers supported by emWin (https://www.segger.com/products/user-interface/emwin/technology/display-drivers/) and u8g2, but the difference is: For u8g2 all displays are tested and can be purchased as a signle item in the internet.

There are multiple other libs:
 * http://embeddedlightning.com/ugui/
 * https://www.ramtex.dk/
 * https://www.segger.com/products/user-interface/emwin/

Comparison is interesting and I am willing to add new features, but my own resources are still limited.

closed 2020-01-25 by olikraus