olikraus/u8g2 #1082: Advise, please: Large Icons in fonts or bitmaps ?

JeffCalwood, 2020-01-01:
Oliver, your advice please: I would like to use text, some (text sized) emojis (I am getting strings from the internet) and some large icons.
- I believe glyphs in a font have a typical size, but the bounding box is not mandatory for all glyphs. Maybe I am wrong. But does this mean I could store a few large icons inside my font ? I could then perform all my drawing using text features, and also I would take advantage of your runlength encoding. How would I go about doing the above ? Or doesn't it make sense ?
- Related, probably a subset of the above, how would you suggest I add a few emoji icons into a font ? Are you aware of a tool with good copy&paste between fonts without having to draw them in, say, Fony manually.
Thanks a lot !

olikraus, 2020-01-01:
> But does this mean I could store a few large icons inside my font ? 

This can be done. There is only one limitation: After compression, the size of one image must not exceed 256 bytes. It is a difficult to predict this, but usually a 64x64 image should be ok. Larger images are also ok, if there are not so many pixel set on the image.

1. The workflow is this: Convert your images to .png file format. 
2. Convert the .png files to a .bdf font file (png2bdf.exe, see here: https://github.com/olikraus/u8g2/tree/master/tools/font/png2bdf)
3. Convert the .bdf font file to the internal u8g2 format (bdfconv.exe, here: https://github.com/olikraus/u8g2/tree/master/tools/font/bdfconv)

> Related, probably a subset of the above, how would you suggest I add a few emoji icons into a font ? Are you aware of a tool with good copy&paste between fonts without having to draw them in, say, Fony manually.

I have never used fontforge, but I could image, that you can use this to copy bitmaps. At the end, I often ended up drawing my own icons. 

I mentioned the unicode emoticons already in #1079, but did you also had a look at my own emoticons:
https://github.com/olikraus/u8g2/wiki/fntgrpu8g#emoticons21

olikraus, 2020-01-25:
no further feedback, closing...

closed 2020-01-25 by olikraus