olikraus/u8g2 #1084: STM32F103C8T6 - ST7920 128x64 GLCD and U8G2 communication problem when I connect CS pin

logiclayer, 2020-01-05:
Hello,
I'm very new to STM32CubeIDE. First of all, I've tested your library in Arduino IDE with STM32103C8T6 and it works fine. Then decided to move to STM32CubeIDE and researched templates but did not work. After a long time I see that when I connect RS(CS) pin on LCD to GPIO pin on STM32 which is defined as LCD_CS then it display nothing. But when I disconnect it then it starts to work. Why is this ? I'm a beginner level of STM and C.
My code is ;
`uint8_t u8g2_gpio_and_delay_stm32(U8X8_UNUSED u8x8_t *u8x8,
		U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int,
		U8X8_UNUSED void *arg_ptr) {
	switch (msg) {
	//Initialize SPI peripheral
	case U8X8_MSG_GPIO_AND_DELAY_INIT:
		/* HAL initialization contains all what we need so we can skip this part. */

		break;

		//Function which implements a delay, arg_int contains the amount of ms
	case U8X8_MSG_DELAY_MILLI:
		HAL_Delay(arg_int);

		break;
		//Function which delays 10us
	case U8X8_MSG_DELAY_10MICRO:
		for (uint16_t n = 0; n < 320; n++) {
			__NOP();
		}

		break;
		//Function which delays 100ns
	case U8X8_MSG_DELAY_100NANO:
		__NOP();

		break;
		//Function to define the logic level of the clockline
	case U8X8_MSG_GPIO_SPI_CLOCK:
		if (arg_int)
			HAL_GPIO_WritePin(SCK_GPIO_Port, SCK_Pin, RESET);
		else
			HAL_GPIO_WritePin(SCK_GPIO_Port, SCK_Pin, SET);

		break;
		//Function to define the logic level of the data line to the display
	case U8X8_MSG_GPIO_SPI_DATA:
		if (arg_int)
			HAL_GPIO_WritePin(MOSI_GPIO_Port, MOSI_Pin, SET);
		else
			HAL_GPIO_WritePin(MOSI_GPIO_Port, MOSI_Pin, RESET);

		break;
		// Function to define the logic level of the CS line
	case U8X8_MSG_GPIO_CS:
		if (arg_int)
			HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, RESET);
		else
			HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, SET);

		break;
		//Function to define the logic level of the Data/ Command line
	case U8X8_MSG_GPIO_DC:
//			if (arg_int) HAL_GPIO_WritePin(CD_LCD_PORT, CD_LCD_PIN, SET);
//			else HAL_GPIO_WritePin(CD_LCD_PORT, CD_LCD_PIN, RESET);

		break;
		//Function to define the logic level of the RESET line
	case U8X8_MSG_GPIO_RESET:
		if (arg_int)
			HAL_GPIO_WritePin(LCD_RESET_GPIO_Port, LCD_RESET_Pin, SET);
		else
			HAL_GPIO_WritePin(LCD_RESET_GPIO_Port, LCD_RESET_Pin, RESET);

		break;
	default:
		return 0; //A message was received which is not implemented, return 0 to indicate an error
	}

	return 1; // command processed successfully.
}

u8g2_Setup_st7920_s_128x64_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_stm32); /* init u8g2 contetx */

	u8g2_InitDisplay(&u8g2); // send init sequence to the display, display is in sleep mode after this,
	u8g2_SetPowerSave(&u8g2, 0); // wake up display`

olikraus, 2020-01-05:
Looks like SET and RESET need to be swapped.

logiclayer, 2020-01-05:
> Looks like SET and RESET need to be swapped.

in this part ? 
`case U8X8_MSG_GPIO_CS:
	if (arg_int)
		HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, RESET);
	else
		HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, SET);

	break;`

olikraus, 2020-01-05:
Yes

logiclayer, 2020-01-05:
> Yes

Yes, it works right now, thanks a lot :)

olikraus, 2020-01-05:
:)

closed 2020-01-06 by logiclayer

Pacimani, 2021-11-28:
Hi, I am working on a similar project. Which call back function should I use to transfer the data? Thanks

olikraus, 2021-11-28:
I think this requires more information.

Pacimani, 2021-11-28:
Hi, thanks for your quick response. What I mean was, since when initializing the LCD requires the delay callback function and a callback function for transmitting data, how can I define/implement one for transmitting data?

olikraus, 2021-11-28:
You only need the gpio and delay function, this callback is also used to transmit the data unless you do this with the byte callback. Did you read the porting guide in the wiki for this project?

Pacimani, 2021-11-29:
Where can I find the gpio function? I was looking for a stm32cubeide guide but I could find it. do you have it somewhere or example codes? thanks

olikraus, 2021-11-29:
This project has been written for Arduino IDE. I don't have any information on any other port.

Pacimani, 2021-11-29:
Ok thanks, as per the initial post, it looks like the codes was meant for stm32cubeide which is what I am using too. Maybe if you read the first post you will understand what I am looking for.

olikraus, 2021-11-30:
I never worked with stm32cubeide 🤔

Pacimani, 2021-11-30:
Thank you very much for your response.