olikraus/u8g2 #1085: Refresh/scan rate is very slow.

AdinAck, 2020-01-05:
I'm not sure if I'm doing this completely wrong or something but I cannot figure out how to make this screen draw more than ~5 frames per second. Just drawing one pixel takes about .75 seconds.

I'm using an Adafruit Grand Central (SAMD51).
The display is an SSD1322 256x64.
The communication protocol is 4 wire SPI.

Here is a video: [link](https://youtu.be/-3UYK496GRo)
and here is my code:

#include <U8g2lib.h>
#include <Wire.h>
#include <SPI.h>

U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* cs=*/ 6, /* dc=*/ 9, /* reset=*/ 5);  // Enable U8G2_16BIT in u8g2.h

int a = 0;

void setup() {
  u8g2.begin();
  u8g2.clearBuffer();
  u8g2.setCursor(0,0);
  u8g2.setFont(u8g2_font_helvB18_tr);
}

void loop() {
  u8g2.clearBuffer();
  u8g2.drawBox(128-25,32-25,50,50);
  u8g2.sendBuffer();
  u8g2.clearBuffer();
  u8g2.sendBuffer();
}

Thank you for any help!

AdinAck, 2020-01-06:
update: tried 8080 to see if that would work better, however, 8080 constructor does not exist for SSD1322 ;(

olikraus, 2020-01-06:
You should try HW_SPI. 
SW SPI will always be very slow.

AdinAck, 2020-01-07:
Oh hey man, thanks for helping out!
So I did try that, it doesn't make any sense to me because the constructor for that has no clock or data pins.

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI(rotation, cs, dc [, reset]) [full framebuffer, size = 2048 bytes]

Is that a typo?

Thanks!

AdinAck, 2020-01-07:
Update, just tried this:

U8G2_SSD1322_NHD_256X64_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 52, /* data=*/ 51, /* cs=*/ 2, /* dc=*/ 3, /* reset=*/ 4);

When verifying it says that function doesn't exist.

olikraus, 2020-01-07:
Clock and data pins are fixed by your hardware, so there's no need to specify them in the ino script. See the description of your board for the location of the SPI pins.

AdinAck, 2020-01-07:
Ah! Let me guess, SW stands for software and HW stands for hardware.
Awesome, I'll try it as soon as I can!

Thanks!

olikraus, 2020-01-07:
yes, MOSI (SPI data) pin seems to be # 50 and SCK (SPI clock) pin is # 52 on the Adafruit Grand Central board.

AdinAck, 2020-01-08:
Works SUPER well! However... it appears that u8g2 is sending slightly offset coordinates to the screen? Here is a video: [link](https://youtu.be/3WYvGG6BlhM)

And here is the code:
#include <U8g2lib.h>
#include <Wire.h>
#include <SPI.h>

U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 2, /* dc=*/ 4, /* reset=*/ 3);  // Enable U8G2_16BIT in u8g2.h

int a = 128;
int b = 1;

void setup() {
  u8g2.begin();
  u8g2.clearBuffer();
  u8g2.setCursor(-20,0);
  u8g2.setFont(u8g2_font_helvB18_tr);
}

void loop() {
  u8g2.clearBuffer();
  u8g2.drawBox(a,32-25,50,50);
  u8g2.sendBuffer();
  a = a + b;
  if (a >= 256-26 or a <= 26){
    if (b == 1){
      b = -1;
    }else{
      b = 1;
    }
  }
}

Just a square going back and forth, It should touch both edges (with a pixel to spare). But it seems that some weird x position glitches are going on?

Thank you so much!

olikraus, 2020-01-08:
Did you enable 16 bit mode in u8g2.h as mentioned in the constructor comment? See also the u8g2 FAQ.

olikraus, 2020-01-25:
no further feedback, closing...

closed 2020-01-25 by olikraus

AdinAck, 2020-02-05:
Sorry, was busy. That did the trick! Thank you so much!

olikraus, 2020-02-05:
:-)