olikraus/u8g2 #1086: Operator Error and Lack of Skill with SED1330, G321D LCD

vapecloudbear, 2020-01-08:
Good day,
I have a G321D display with an integrated SED1330 Controller. I am trying to use an Arduino Uno to run this display, but I don't know where to begin. My first struggle is finding a negative voltage power source for the V0 and VLC pins, but after I overcome this, how would I actually wire this up to the Arduino to run it? And after that, would the SED1330 https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sed1330-240x128
or the https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ra8835-320x240
possibly work for this? Thank you for what you do! Your work is awesome.

Display info here: https://www.digchip.com/datasheets/parts/datasheet/798/G321D-pdf.php

Detailed datasheet here: http://compsys1.com/support/docs/g321.pdf

Controller datasheet here: https://www.lcd-module.de/eng/pdf/zubehoer/sed1330.pdf

olikraus, 2020-01-08:
You display is a 320x200 variant, so we probably have to adjust the u8g2 software a little bit (shouln't be a problem).
You can start testing with the existing drivers. Maybe the RA8835 320x240 will already work, but you could also test the sed1330 240x128. The output might be wrong, but if you send a picture, I should be able to create a new constructor for your device.

The G321D probably does not contain a negative DC/DC converter.

To generate the negative voltage from a 5V source you probably need a dc/dc step up converter, which generates .-24 volt (maybe -15 are also sufficient).
Something like this: https://recom-power.com/de/products/dc-dc-converters/rec-p-RB-0524D.html?4

Not sure whether ebay is a reliable source, but there is also such a converter:
https://www.ebay.com/itm/Positive-Negative-Dual-Output-DC-DC-Step-up-Boost-Converter-5V-9V-12V-24V/263923992688

vapecloudbear, 2020-01-08:
Thank you for the quick response. I will gladly test and send photos once I get a working power supply. I requested a sample from recon, hopefully they will approve it. When I try to test with existing ones, which pins will I connect to Arduino and where? I’m lost with the read, write, and SEL pins. I appreciate the help.

vapecloudbear, 2020-01-08:
[image]

olikraus, 2020-01-08:
You will need a 5V Arduino board (e.g. Arduino Uno).
There are no restrictions for the pins, you can choose any GPIO of your board. Later, you need to tell u8g2 which pin is connected to which function fo the display.
Best is to write down a table, where you describe which display pin is connected to which Arduino Pin: https://github.com/olikraus/u8g2/wiki/setup_tutorial#identify-digital-lines

There is also a nice video here: https://www.youtube.com/watch?v=4CD8ERaylmY

That person first tried randomly to use u8glib (predecessor of u8g2) until he systematically wrote down the pin connections (which I always suggest).

vapecloudbear, 2023-05-18:
Hello again @olikraus. I am finally back trying again to get this to work. I am still very confused on the pin matching. Based on the screenshot above, I’ve got the following: lcd 20 to ground. 19 to -24vdc. 18 to a 10k pot connected to ground and -24vdc. 17 to ground. 16 to 5vdc. 8-15 to 8,9,10,11,4,5,6,7 on the uno per the driver. I have 4 and 5 connected ground. 1. /res to reset 16. My confusion is with lcd pins 2,3,6, and 7. I think 3 /wr  goes to enable (arduino 17). Pin 6 /cs goes to cs 14 on the arduino. So what about 2 and 7? Is this correct so far? Thanks. I’m trying to use the sed1330_240x128 driver. Thanks!

vapecloudbear, 2023-05-18:
I think I answered my question. 2 goes to 5vdc. And 7/A0 goes to arduino 15. It tries to work
[image]

olikraus, 2023-05-19:
Looks good, but I assume you need to use the picture loop.

vapecloudbear, 2023-05-19:
Just reuploaded the sketch. This is using example->U8g2->page_buffer->U8g2Logo. I have also tried other examples in the page buffer folder. It seems to draw the first 8 pixels left to right, then everything is garbled after that. Thanks for your help.

olikraus, 2023-05-19:
Which constructor did you use?
Can you post the code (constructor/setup()/main())?

olikraus, 2023-05-19:
Your display is a 320x200 display, correct? So it is anyways not supported and we need to create a new constructor, right?
Additionally: Did you enable 16 bit mode?

vapecloudbear, 2023-05-19:
I used the constructor: `U8G2_SED1330_240X128_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 17, /*cs=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect /RD = E with +5V, enable is /WR = RW, FG with GND, 14=Uno Pin A0` and also tried the RA8835_NHD_240x128 with the same results. 
code:
`void setup(void) {
  u8g2.begin();
}

void drawLogo(void)
{
  uint8_t mdy = 0;
  if ( u8g2.getDisplayHeight() < 59 )
    mdy = 5;
  

    u8g2.setFontMode(1);	// Transparent
    u8g2.setDrawColor(1);
#ifdef MINI_LOGO

    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb16_mf);
    u8g2.drawStr(0, 22, "U");
    
    u8g2.setFontDirection(1);
    u8g2.setFont(u8g2_font_inb19_mn);
    u8g2.drawStr(14,8,"8");
    
    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb16_mf);
    u8g2.drawStr(36,22,"g");
    u8g2.drawStr(48,22,"\xb2");
    
    u8g2.drawHLine(2, 25, 34);
    u8g2.drawHLine(3, 26, 34);
    u8g2.drawVLine(32, 22, 12);
    u8g2.drawVLine(33, 23, 12);
#else

    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb24_mf);
    u8g2.drawStr(0, 30-mdy, "U");
    
    u8g2.setFontDirection(1);
    u8g2.setFont(u8g2_font_inb30_mn);
    u8g2.drawStr(21,8-mdy,"8");
        
    u8g2.setFontDirection(0);
    u8g2.setFont(u8g2_font_inb24_mf);
    u8g2.drawStr(51,30-mdy,"g");
    u8g2.drawStr(67,30-mdy,"\xb2");
    
    u8g2.drawHLine(2, 35-mdy, 47);
    u8g2.drawHLine(3, 36-mdy, 47);
    u8g2.drawVLine(45, 32-mdy, 12);
    u8g2.drawVLine(46, 33-mdy, 12);
    
#endif

}

void drawURL(void)
{
#ifndef MINI_LOGO
  u8g2.setFont(u8g2_font_4x6_tr);
  if ( u8g2.getDisplayHeight() < 59 )
  {
    u8g2.drawStr(89,20-5,"github.com");
    u8g2.drawStr(73,29-5,"/olikraus/u8g2");
  }
  else
  {
    u8g2.drawStr(1,54,"github.com/olikraus/u8g2");
  }
#endif
}

void loop(void) {
  u8g2.firstPage();
  do {
    drawLogo();
    drawURL();
  } while ( u8g2.nextPage() );
  delay(4000);
}
`

vapecloudbear, 2023-05-19:
To answer the other question, yes. This is a 320x200 and needed a new constructor. You were assisting me here and on the arduino forums 3 years ago. Life got too crazy and I had to step away from my project for a while. If there is a way I can assist you in your efforts, please let me know. :-) Thanks again

vapecloudbear, 2023-05-19:
Changing to 16Bit mode results in no visual changes at all.

olikraus, 2023-05-21:
I have created a new beta release 2.34.18 with the following constructors

``` C++
U8G2_SED1330_320X200_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 17, /*cs=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect /RD = E with +5V, enable is /WR = RW, FG with GND, 14=Uno Pin A0, enable U8g2 16Bit Mode 
U8G2_SED1330_320X200_1_6800 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,  /*enable=*/ 17, /*cs=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // A0 is dc pin, /WR = RW = GND, enable is /RD = E, enable U8g2 16Bit Mode 
```

Can you test this constructor. What will be shown on the display?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

vapecloudbear, 2023-05-23:
U8G2 logo sketch:

vapecloudbear, 2023-05-24:
CORRECTION! Everything appears to work now. I pulled off every wire and reconnected. So far, the logo, hello world, and graphics test sketches have all loaded and rendered properly. Thank you and sorry for the miscommunication. Now I am trying to figure out why I am getting a screen flicker.

vapecloudbear, 2023-05-24:
/

vapecloudbear, 2023-05-24:
[image]

olikraus, 2023-05-28:
Sorry also for my delay. Nice pictures.
So this means, the new constructor works as expected?

vapecloudbear, 2023-05-28:
Yes, the driver commands work as expected. However, what I thought was just a wiring issue on my part turned out not to be. So every time I unplug the Zuni and plug it back in, the screen doesn’t initialize properly. I have to unplug the Read with from 5vdc, let it boot. Then reset the arduino, then plug the enable wire back in, then reset once more.  I have 4 of these displays and have tried it on two of them, both driver the same results. But in reading the docs for the SED1330 driver, I can’t tell if the read pin needs to be pulled down to ground for the initialization or reset of the screen. Once I get the graphics to display correctly, I can change or reupload code, and everything will continue to work until I unplug the arduino.

olikraus, 2023-05-29:
What kind of Arduino board did you use? Not sure whether you mentioned this before. But if it is a 3.3V board, then you might need level shifters for the display. 

You could increase this in your local copy of u8g2:
https://github.com/olikraus/u8g2/blob/cbb28b9ffd992aea74a8be72b514ccd2159cf47b/csrc/u8x8_d_sed1330.c#L459-L460

Maybe increase both values to 20 ms.

vapecloudbear, 2023-05-30:
5v arduino Uno. Setting both values to 20 and recompiling/uploading resulted in no changes.

vapecloudbear, 2023-05-31:
Now that I better understand the driver, I took the liberty of making a few changes to match it up with the datasheet, and now the screen initializes consistently. Below is a screenshot of the changes made, and the modified code to update your repo if you choose. (Values taken from page 26 of the Datasheet) I did notice that the values in the current driver were setting the lines to 240 instead of 200.
[image]

Code:
`  U8X8_A(0x087),				/* no idea here... WF (topmost bit) is set to one because it is suggested in the datasheet, lowest 3 bits refer to text mode only */
  U8X8_A(0x007),				/* FY: height of a char+1, does not matter here (hopefully), because we use graphics mode only */
  U8X8_A(0x027),	/* 40-1 */		/* C/R: this could be the number of horizontal bytes - 1 (Value confirmed with app notes p41) */
  U8X8_A(0x02F),				/* TC/R: According to app notes fOSC=6Mhz fFF=70Hz --> TC/R = 74d*/
  U8X8_A(0x0c7),				/* 0xc7=199, L/F: Lines per frame - 1, probably this is the height of the display - 1 (value confirmed with app notes p41)*/
  U8X8_A(0x028),				/* Low byte of the virtual screen size. (Value confirmed with app notes p41)   */
  U8X8_A(0x000),				/* High byte of the virtual screen size, see also section 9.1.2 */
	
  U8X8_C(0x044),				/* SCROLL */
  U8X8_A(0x000),				/* Screen 1 Start Address (Low) = 00H */
  U8X8_A(0x000),				/* Screen 1 Start Address (High) = 00H */
  U8X8_A(0x0c7),                                /* Screen Layer 1 Number of lines = 200d */ 
  U8X8_A(0x0b0),                                /* Screen 2 Start Address (Low) = 00H */
  U8X8_A(0x005),                                /* Screen 2 Start Address (High) = 00H */
  U8X8_A(0x0c7),                                /* Screen Layer 2 Number of lines = 200d */
  U8X8_C(0x04C),                                /* Cursor Direction 0,0 Shift Direction = Right */`

olikraus, 2023-06-06:
Thanks for the feedback.
So you say, your modified green code works better?

olikraus, 2023-06-06:
I am a little bit confused. According to the datasheet from here: https://www.lcd-module.de/eng/pdf/zubehoer/sed1330.pdf
The scroll command has a total of 11 bytes. So I kept the 0 bytes here:

https://github.com/olikraus/u8g2/blob/a143183f0d117d57572a3646567a78900b0de75d/csrc/u8x8_d_sed1330.c#L476-L507

vapecloudbear, 2023-06-06:
That works. Thanks again for all of your help after 3 years. :-) You rock!

closed 2023-06-06 by vapecloudbear