olikraus/u8g2 #1087: Using bdfconv to make custom fonts - "error, glyph too large, encoding=34 cnt=303"

deezums, 2020-01-08:
Hello again!

I am trying to use Fony and bdfconv to make a 96x96 pixel icon font for use with icon menu. For some reason, it seems Fony can't export larger than 64 pixel bitmaps, so I am attempting to use 64x64 pixel font/icons. 

I have sucessfully imported bitmaps into fony and have a font saved in BDF format. When I import the font into Fony again everything looks correct. I then try using bdfconv to make the font file, but I get the error message. 

C:\...\Icons>bdfconv -v -f 1 -m "32-42" -n Font_Ico -o font.c font.bdf
Parse File font.bdf: 256 glyph(s) found
Map: map_cmd_list='32-42'
Map: exclude=0 from=32/$20 to=42/$2a map=32/$20
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=0, w=64, h=64
CalculateMaxBBX: Encodings x=32, y=32, w=32, h=33
bf_CalculateMinMaxDWidth: dx_min=50, dx_max=65
bf_CalculateMinMaxDWidth: x_min=0, x_max=0
bf_CalculateMaxBitFieldSize: bbx.x=1, bbx.y=3, bbx.w=7, bbx.h=7, dwidth=8
RLE Compress: best zero bits 6, one bits 6, total bit size 15768
RLE Compress: Font code generation, selected glyphs=11, total glyphs=256
RLE Compress: Error, glyph too large, encoding=34 cnt=303

I know there are 64x64 pixel fonts (iconic 8x) so I am wondering what I am doing wrong or what I can change to make this work properly?

Thank you!

olikraus, 2020-01-08:
After compression the glyph must fit into a 256 buffer. This is not the case with your icon. It requires 303 bytes after compression. Compression will work better on simpler images.

deezums, 2020-01-08:
Ah, that makes perfect sense. I finally listened to what the errors were telling me and managed to "see" glyph 33 using u8g2 by clipping the .bdf file. At least I know I am doing everything else mostly correct now. 

The software "FontForge" works much better than Fony with >64 pixel fonts, too, although it's a screaming pile of pain in the butt to use on windows (will only export a BDF once before crashing) but beggars can't be choosers!

Thank you!

closed 2020-01-08 by deezums

olikraus, 2020-01-08:
:)