olikraus/u8g2 #1088: Open-Smart ST7567 -Support buggy

AndreasFFM, 2020-01-12:
First, many thanks for creating this lib!

I'm using an ["OPEN-SMART 3.3V 2.6 Inch 128x64 Serial SPI Monochrome LCD Breakout Board Module with Backlight for Arduino Nano Pro Mini"](https://www.dx.com/p/opensmart-33v-26-inch-128x64-serial-spi-monochrome-lcd-breakout-board-module-with-backlight-for-arduino-nano-pro-mini-2710499.html) on an Arduino Due with u8g2 2.16.14 using arduino IDE 1.8.9 and trying to run the weather.ino demo with SW_SPI.

The U8G2_ST7567_PI_132X64_1_4W_SW_SPI constructor sets the screen to black, then nothing happens. Adding a u8g2.setContrast(50); to the setup will turn the black screen back to normal, and will display changing symbols only in the top 8 pixels.

U8G2_ST7567_JLX12864_1_4W_SW_SPI,U8G2_ST7567_ENH_DG128064_1_4W_SW_SPI and U8G2_ST7567_ENH_DG128064I_1_4W_SW_SPI only showed a white screen.

I took U8G2_ST7567_JLX12864_1_4W_SW_SPI and mode 3 modifications:
1) add u8g2.setContrast(50); in the demo sketch setup
2) In u8x8_st7567_enh_dg128064_init_seq[],
replace
  U8X8_C(0x023),		                /* v0 voltage resistor ratio */
  U8X8_CA(0x081, 200>>2),		/* set contrast, contrast value*/
with
  U8X8_C(0x026),		                /* v0 voltage resistor ratio */
  U8X8_CA(0x081, 50>>2),		/* set contrast, contrast value*/
(taken from the 132X64 ini sequence, and the value from the setContrast call. Unfortunately the setContrast call is still additionally required, it seems the contrast gets lost somewhere)
This makes the display show a black screen first, then the top 8 lines of graphics.
3) In u8g2_m_16_8_1
replace
  static uint8_t buf[128];
  *page_cnt = 1;
with
  static uint8_t buf[128*8];
  *page_cnt = 8;
This will show the full 8*8 lines instead of only the first 1*8 lines.

I don't understand your libaries structure good enough to create an official patch from that, but for me it lets the weather demo run nicely.

olikraus, 2020-01-12:
Thanks. I will add the display.

olikraus, 2020-01-12:
Hmm... this does not fit, You have used U8G2_ST7567_JLX12864_1_4W_SW_SPI, but you modified u8x8_st7567_enh_dg128064_init_seq.

Well, let's see, I will create a new device and maybe you can check whether it works.

olikraus, 2020-01-12:
The new constructor looks like this:
U8G2_ST7567_OS12864_1_4W_SW_SPI
U8G2_ST7567_OS12864_1_4W_HW_SPI

Can you test the new device?

If there is something wrong, you can modify your local code here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7567.c#L772

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2020-01-12:
New test release is 2.27.4.

AndreasFFM, 2020-01-12:
Wow, thats a fast response, thank you :-)

> Hmm... this does not fit, You have used U8G2_ST7567_JLX12864_1_4W_SW_SPI, but you modified 
> u8x8_st7567_enh_dg128064_init_seq.
You are right. I modified all of them at some point of testing, and it seems i used the wrong one for the final contrast test.

I downloaded and installed the master.zip
Using the new constructor U8G2_ST7567_OS12864_1_4W_SW_SPI  i now get the display initialized with correct contrast (without additionally calling setContrast). However, it still only displays the top 8 pixel rows.

If i change the u8g2_Setup_st7567_os12864_1 in u8g2_d_setup.c to use
u8g2_m_16_8_f  (1024 Bytes, 8 pages) 
instead of 
u8g2_m_16_8_1 (128 Bytes, 1 pages)
the full display works correctly again.

The display has 128x64 pixels, so the buffer should have 128x64/8 bytes of size not 128 bytes, or do i misunderstand how the buffer is used?

olikraus, 2020-01-12:
Oh, you need to use the _F_ version of the constructor. Just replace _1_ with _F_ in the constructor name.

AndreasFFM, 2020-01-13:
Ah, i see. Using U8G2_ST7567_OS12864_F_4W_SW_SPI constructor (and the correct pins), the weather demo works out of the box.  Thank you!

I think the "default constructors" included in comments in the demos could use some overhaul. The "fullbuffer\weather"-demo for example only includes the _1_ constructors.

closed 2020-01-19 by olikraus