olikraus/u8g2 #1089: Pin re mapping

Sozuak, 2020-01-12:
I have an arduino based  board already designed and completed to work another library and display. I want to use some of them with T6963 with 240x64 (instead of 7108 and 128x64) .  So pins dedicated for display  from my board  did not fit to u8g2lib pins. Is there any definition file to exchange the pins to fit  to my display.

olikraus, 2020-01-12:
Any pins can be used. Just name them on during constructor call.

Sozuak, 2020-01-13:
You mean modifying  this line :

`/U8G2_T6963_128X64_F_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable/wr=*/ 17, /*cs/ce=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect RD with +5V, FS0 and FS1 with GND`

olikraus, 2020-01-13:
Yes, Numbers need to be changed according to your wiring.

Sozuak, 2020-01-14:
Ok I understand, thanks. I got the new numbers.

closed 2020-01-14 by Sozuak

olikraus, 2020-01-14:
:)