olikraus/u8g2 #1090: Could you add the support SSD1207 128 x 96?

hehao9051, 2020-01-13:
Because i don't know how to add the support new oled, could you add  the support SSD1207 128 x 96? the attach is the datasheet, thanks.
[M02289_VGM128096A4W10_Y02.pdf](https://github.com/olikraus/u8g2/files/4052919/M02289_VGM128096A4W10_Y02.pdf)

olikraus, 2020-01-14:
```
//U8G2_SSD1327_VISIONOX_128X96_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_SSD1327_VISIONOX_128X96_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```
I have created beta release 2.27.5, which includes the above constructors.
Can you run the u8g2 page mode flipmode example? It is all ok if the border frame should never changes.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

hehao9051, 2020-01-15:
I add the support for yesterday refer your code and work well in my board. Thanks your work, i will test i t in my board(only the c version, because i have no Arduino).

hehao9051, 2020-01-15:
I test flip mode in my board，i don't know wheter it work well please see the video。

hehao9051, 2020-01-15:
actually i have another two question：1、the meaning of tile_width and tile_hight 。2、can i place font file in external flash memory chip，because chip flash is small。

olikraus, 2020-01-15:
> Uploading PhotoArchive0.zip…

I think the link is broken

> the meaning of tile_width and tile_hight 

Almost all displays are organized in 8x8 pixel areas. So a tile is just a quadratic area of 8 by 8 pixels.
If you have the tile position as tx and ty, the pixel position of the upper left corner of the tile is 8*tx and 8*ty. 

> can i place font file in external flash memory chip，because chip flash is small

U8g2 will access the font via a pointer. If the flash is memory mapped into the address space of your controller, then this shouldn't be a problem. If you have something like a SPI flash, then you need to load the font into RAM first and pass the address of the font in RAM to setFont.

hehao9051, 2020-01-15:
OK，thanks。see the new zip file。
[PhotoArchive.zip](https://github.com/olikraus/u8g2/files/4065555/PhotoArchive.zip)

olikraus, 2020-01-15:
Wow, already amazing.

Can you adjust the second value of line 912 in u8x8_d_ssd1327?

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L912

Instead of 0x020 is should 0x0 only. Will this work? 
The 0xa2 command specifies the y offset (the position of the top pixel row in the display controller RAM) and the value 0x020 is the offset itself.  However this will change if the display is rotated.

hehao9051, 2020-01-15:
OK，i will try it tomorrow and tell you the result。

hehao9051, 2020-01-16:
static const uint8_t u8x8_d_ssd1327_128x96_flip1_seq[] = {
  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
  U8X8_CA(0x0a2, 0x000),		/* display offset, shift mapping ram counter */
  U8X8_CA(0x0a0, 0x042),		/* remap configuration */
  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */
};
please see the video
[cccdba1c3e672ad39c5a63121406b89b.zip](https://github.com/olikraus/u8g2/files/4069305/cccdba1c3e672ad39c5a63121406b89b.zip)

olikraus, 2020-01-16:
hm... ok... this doesn't help. Are you sure you have recompiled everything?
Did you observe any change when modifiying the a2 argument?

hehao9051, 2020-01-17:
i see our driver code, when we want to rorate, we set A2 = 0x060, please look the attach doc.
[diff.docx](https://github.com/olikraus/u8g2/files/4074802/diff.docx)

olikraus, 2020-01-18:
Amazing, great work. So you mean to say that the value 0x060 works best, right?
I have updated the code accordingly. 
Thanks a lot for your contribution and testing effort.

hehao9051, 2020-01-18:
yes，thanks you too

closed 2020-01-19 by olikraus