olikraus/u8g2 #1091: Issue clearing a line of text with some fonts

rickyelqasem, 2020-01-14:
I want to use u8g2.drawStr(0,15,"               ");   to clear that area for different text. I noticed that the spaces only show up in very few fonts. u8g2_font_8x13B_mf is a font that works but is too small for me. As a result the next input of text merges with the previous values.   I dont want to clear the whole screen just that line of text. What am I doing wrong? thx

rickyelqasem, 2020-01-14:
Ok I worked out drawing a blank box fixes the issue. Now I noticed that on that like the text wraps back around to position 0 if its too long for the screen. Anyway to stop that?

olikraus, 2020-01-14:
Most important: You need to set the font mode transparency to false (0): https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode

> u8g2_font_8x13B_mf is a font that works but is too small for me.

Indeed you have to to search for a font where the second last char is an "m" (monospace).
A complete list of those fonts is here: https://github.com/olikraus/u8g2/wiki/fntlistmono
I think you should be able to find a suitable font.
If not. Additional "m" fonts could be created also.

> Now I noticed that on that like the text wraps back around to position 0 if its too long for the screen. Anyway to stop that?

You need to set the u8g2 16 bit mode: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L163
Wrapping will still happen, but not after 2^8 but after 2^16.

olikraus, 2020-01-25:
no further feedback, closing...

closed 2020-01-25 by olikraus