olikraus/u8g2 #1092: D2 pin locked on Heltec WiFi Kit8

Delivery2you, 2020-01-17:
Hey there! I am using u8g2 library and all works well, but since I have a function which constantly updates the screen in the loop based on the pin state and using the following constructor definition

//U8g2 Contructor
U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 16, /* clock=*/ 5, /* data=*/ 4);

It appears my D2(GPIO0) pin is always on, and I have connected it to some external switch.
Is there any possibility to release that pin?
When I mark my Display function in the loop as a comment, I have no issues with that pin.

olikraus, 2020-01-17:
D2 Pin doesn't seem to be related to u8g2.

Delivery2you, 2020-01-18:
Thank you for the response. You were correct, nothing to do with the pin. I had a DHT sensor wrongfully connected to it.

olikraus, 2020-01-18:
thanks for the feedback, i will close this ticket. please reopen if there are any other issues.

closed 2020-01-18 by olikraus