olikraus/u8g2 #1093: Add Nokia font

marmotton, 2020-01-19:
Hello,
I converted the nokiafc22.ttf font by Zeh Fernando to use it with a Nokia 3310 display :)
I used these two commands:
```
./otf2bdf -r 72 -p 8 -o nokiafc22.bdf nokiafc22.ttf
./bdfconv -f 1 -m '32-255' -n u8g2_font_nokiafc22_tf -o u8g2_font_nokiafc22_tf.c nokiafc22.bdf
```
The licensing is not clear, the .txt file claims that we can distribute it freely (but not sell it), the author is not sure whether there is a Nokia copyright. If it sounds ok, can you add it to the library ?

[u8g2_nokiafc22.zip](https://github.com/olikraus/u8g2/files/4082656/u8g2_nokiafc22.zip)

olikraus, 2020-01-19:
Is there any official web page for this font?

marmotton, 2020-01-20:
This looks like the official site, some interesting comments: https://www.dafont.com/font-comment.php?file=nokia_cellphone

I didn't find anything related to this font on his personal website.

olikraus, 2020-01-20:
Sounds ok, thanks

marmotton, 2020-01-21:
Great, thank you!

olikraus, 2020-01-21:
It will be part of the next released:
https://github.com/olikraus/u8g2/wiki/fntgrpdafont

I added also some other fonts from dafont.com

marmotton, 2020-01-21:
Looks good, thanks for taking great care of this library :)

closed 2020-01-21 by olikraus