olikraus/u8g2 #1094: Need help to use thes lib with Heltec CubeCell

joba-1, 2020-01-20:
Hi Oli,

I have a SH1106 display that for some reason does not work with the SSD1306 driver provided with the heltec arduino core for the CubeCell LoRa node (https://github.com/HelTecAutomation/ASR650x-Arduino).

That is why I tried your lib u8g2, but after some small adaptions to get it to compile I get linker errors due to rom overflow (.rodata will not fit...), even with the 8x8 HelloWorld example. With the full and page buffer versions I even get ram overflows. Unfortunately the messages don't tell by how much.

The CubeCell has 128KB flash and 16KB sram. Not a lot, but then it has power consumption of only ~9µA (measured) in sleep mode which makes it an ideal board for outdoor battery usage.
  
I'm ok with just some text output on the display, but LoRaWAN and sensor code needs space, too. A simple sensor sending via LoRaWAN (and no display) already needs ~60% of rom.

Any idea on how to get rom (and maybe ram) usage of u8g2 down even more? Or should I just forget about it?

joba-1, 2020-01-20:
Full build output

[u8g2-helloworld.log](https://github.com/olikraus/u8g2/files/4087231/u8g2-helloworld.log)

olikraus, 2020-01-20:
There is no "-fdata-sections" in the log. So I think it is a problem with your build environment. 
It is mandatory to fully enable the gcc garbage collector for all sections, not only the functions but especially also for the data sections.

joba-1, 2020-01-20:
Thanks for the hint. 
The CubeCell is very new and has a lot of signs of immaturity (beginning with the version number 0.0.4...)
But the examples build ok.
Do you mean this is needed for your lib?

olikraus, 2020-01-20:
> Do you mean this is needed for your lib?

Yes, it is required for u8g2.

joba-1, 2020-01-20:
Ok, I‘ll try to play around with this.

joba-1, 2020-01-21:
Ok, you were right. Now it compiles and flashes. Thank you!

Close to success, I "just" need to find out what kind of display I got. 
The SH1106 HW I2C constructor I thought would be it does not show anything.
I need to provide SCL and SDA with a SW constructor.
So far u8x8 HelloWorld only shows "He" and the full_buffer version only the upper half of "Hello V".  Well, you have plenty more of them to try... :)

closed 2020-01-21 by joba-1

olikraus, 2020-01-22:
Maybe also the Arduino HAL is not yet fully implemented: If SW I2C does not display everything, then maybe pinMode does not jet suppot INPUT_PULLUP option.

joba-1, 2020-01-22:
Thanks for your inputs.

CubeCell Arduino HAL certainly is not fully implemented, although they claim "Perfect Arduino-Compatible!". But it seems, they at least work on it and the product is young (november, I think).

Most of the things I had to fix/workaround so far were SPI related compile issues (although I only need I2C).
If you are interested: https://github.com/HelTecAutomation/ASR650x-Arduino/compare/master...joba-1:master

I had to switch from core version 0.0.4 to git because this included INPUT_PULLUP 12 days ago.
It looks hackish to me, no idea if this is fully equivalent:

https://github.com/HelTecAutomation/ASR650x-Arduino/commit/266b959bee09a2bfb812d13ec18f0252c22d4dd7
```
#define PINMODE_INPUT_PULLUP(pin) ({pinMode(pin,OUTPUT_PULLUP); digitalWrite(pin,HIGH);})
#define PINMODE_INPUT_PULLDOWN(pin) ({pinMode(pin,OUTPUT_PULLDOWN); digitalWrite(pin,LOW);})
```

Not sure what the correct mode would be for an I2C bus which is read and write. 
From those implemented, pinMode(pin, OD_HI) I guess, but the board I use has 10k pullup resistors, so I think a PULLUP mode shouldn't be needed.

Anyways, I tried all avilable constructors and am back at SH1106 with SW I2C as it shows the best result (no garbage patterns, just not complete).

I'll keep on trying...

joba-1, 2020-01-25:
Just to finish this with a nice conclusion: that partial output must have been a breadboard issue.
After exchanging it, the display works just fine with your libs u8x8 or u8g2. 
Plenty of flash and ram space left to pick up some data worth displaying...