olikraus/u8g2 #1095: Oled 64x32 ssd1306, only botton of caracter are drawing

manuprendlair, 2020-01-22:
Hi,

I got this problem:

[image]

My config:

Adafruit feather M0 adalogger.

code:

```
#include <Wire.h>
#include <U8g2lib.h>

U8G2_SSD1306_64X32_1F_1_HW_I2C U8G2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup() {
  Serial.begin(115200);
  Serial.println("\r\nAnalog logger test");
  pinMode(13, OUTPUT);

  U8G2.begin();
  U8G2.clearBuffer();
  U8G2.setFontMode(1);// clear the internal memory
  U8G2.setFont(u8g2_font_cu12_tr);	// choose a suitable font
  U8G2.drawStr(0, 5, "Hello");	 // write something to the internal memory

  U8G2.sendBuffer();					// transfer internal memory to the

  delay(2000);
}
```
I check with all U8G2_SSD1306_64X32 .. no one work!
Spent many time to search on Internet, but I realy don't know what's happend

I greatly appreciate some help

Thanks

olikraus, 2020-01-22:
Use `U8G2_SSD1306_64X32_1F_F_HW_I2C`!

manuprendlair, 2020-01-22:
oK, in fact my display was sell for 64x32 but it is 64x48.
All work.
This issue is not an issue and could be close.

Thanks

closed 2020-01-22 by manuprendlair