olikraus/u8g2 #1096: serial port problem

planer-pro, 2020-01-23:
Hi i use your lib in arduino. Lib is really great! Use userinterface function. In this mode arduino serial port is not working, it will be great suprise for me)) Is it possibility work userinterface function with serial port arduino together?

olikraus, 2020-01-23:
Which board do you use? Which pins for you use for buttons?

planer-pro, 2020-01-24:
I use mega2560 r3
As i inspected serial port didn't work because every `inputPass_selection = u8g2.userInterfaceMessage("SYSTEM BLOCKED", "Enter password", "", " ok ");` or `inputPass_selection = u8g2.userInterfaceSelectionList("INPUT PASS", inputPass_selection, number_list);` halting main loop function

olikraus, 2020-01-24:
> It's a pity, what i must refuse nice library.

It is not the goal of this lib to provide a user interface. Functions like the userInterfaceMessage are only there to demonstrate the graphics capabilities.

Instead you probably may want to use https://github.com/neu-rah/ArduinoMenu which is a menu system based on an existing low level graphics/character output system (such as u8g2).

planer-pro, 2020-01-24:
Ok, Thanks!

closed 2020-01-25 by olikraus