olikraus/u8g2 #1097: Missing SSD1306 72X40_ER in U8g2lib.h (Arduino)

msommerx, 2020-01-24:
Hi, I noticed that the display you refere to [https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-72x40_er-1](url) is missing in the U8g2lib.h.
Can you please check?
Thanks, Martin

olikraus, 2020-01-24:
It is not yet released, but is part of the latest beta. Github is a little bit difficult. While there is a good release management for the software, the wiki always shows the latest changes.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2020-01-24:
https://github.com/olikraus/u8g2/issues/1097

closed 2020-01-24 by msommerx