olikraus/u8g2 #1098: nrf51822 + 1.5 inch OLED Waveshare SSD1327

iTzZMaestro, 2020-01-26:
Hello,
I'm trying to turn on my OLED Waveshare 128*128 SSD1327 screen using i2c communication with the NRF51 DK development kit but unfortunately my module doesn't turn on.
I read and tried without success the following tutorial https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication to use the u8x8_byte_sw_i2c procedure and my own GPIO and Delay callback function.
My screen is connected to the nrf51 DK as follows:
Screen DIN to P0_30 (SDA)
Scren CLK to P0_7 (SCL)
Screen DC to GND (I had also tested it by plugging it on pin P0_1 of the development kit)
Screen CS to P0_2
Screen RST to P0_3

I've debugged the code with printf to see if the code gets stuck somewhere but it seems to me that there is no problem.

Do you have any suggestions on my functions or connections to make this screen work please

Thanks in advance

`uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
      

  switch(msg)
  {
 
    case U8X8_MSG_GPIO_AND_DELAY_INIT:
    // called once during init phase of u8g2/u8x8
      break; // can be used to setup pins

    case U8X8_MSG_DELAY_NANO:
    // delay arg_int * 1 nano second
      break;   

    case U8X8_MSG_DELAY_100NANO:
    // delay arg_int * 100 nano seconds
      break;

    case U8X8_MSG_DELAY_10MICRO:
   	// delay arg_int * 10 micro seconds
      break;

    case U8X8_MSG_DELAY_MILLI:
  //  printf("U8X8_MSG_DELAY_MILLI MSG : %d\n", msg);			// delay arg_int * 1 milli second
    delay_micro_seconds(arg_int * 1000UL);
      break;

    case U8X8_MSG_DELAY_I2C:
   // printf("U8X8_MSG_DELAY_I2C MSG : %d\n", msg);	
   delay_micro_seconds(arg_int<=2?5:1);			// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    
    case U8X8_MSG_GPIO_CS:
     nrf_gpio_pin_write(P0_2, arg_int);					// CS (chip select) pin: Output level in arg_int
      break;

    case U8X8_MSG_GPIO_DC:
    nrf_gpio_pin_write(P0_1, arg_int);				// DC (data/cmd, A0, register select) pin: Output level in arg_int
      break;

    case U8X8_MSG_GPIO_RESET:
   // printf("U8X8_MSG_GPIO_RESET: %d\n", msg);
    nrf_gpio_pin_write(P0_3, arg_int);			// Reset pin: Output level in arg_int
      break;

    case U8X8_MSG_GPIO_I2C_CLOCK:
  //  printf("U8X8_MSG_GPIO_I2C_CLOCK MSG: %d\n", msg);
      nrf_gpio_pin_write(P0_7, arg_int);	
  // arg_int=0: Output low at I2C clock pin
      break;	// arg_int=1: Input dir with pullup high for I2C clock pin

    case U8X8_MSG_GPIO_I2C_DATA:
  //  printf("U8X8_MSG_GPIO_I2C_DATA: %d\n", msg);
   nrf_gpio_pin_write(P0_30, arg_int);	
   	// arg_int=0: Output low at I2C data pin
      break;							// arg_int=1: Input dir with pullup high for I2C data 
      
    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
      break;

    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
      break;

    case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
      break;

    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
      break;

    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;
  }
  return 1;
}`

Here my setup

`  void setup() {

    u8g2_Setup_ssd1327_i2c_ws_128x128_f(&myScreen, U8G2_R0, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_template);
    //u8g2_SetI2CAddress(&myScreen, 0x3C);
    u8g2_InitDisplay(&myScreen);
    u8g2_SetPowerSave(&myScreen, 0);
    u8g2_ClearBuffer(&myScreen);   
    u8g2_SetFont(&myScreen, u8g2_font_ncenB14_tr);
    u8g2_DrawStr(&myScreen, 1, 18, "U8g2 on nordic");
    u8g2_SendBuffer(&myScreen);
 
}`

olikraus, 2020-01-26:
Code looks ok to me

iTzZMaestro, 2020-01-28:
> Code looks ok to me

Hello
Thank you for your response.
I'm still stuck with my screen not wanting to turn on.
I changed the u8x8_byte_sw_i2c procedure to a specific procedure of my hardware. I still don't understand why my screen stays black. I have analyzed the i2c communication with another simple program and it seems to me that the connection is good with the SSD1327. Indeed, I detect an event when the SSD1327 sends a message. I would like you to give me an opinion on my codes and other tips that will allow me to make this screen work. One last question, how can I ensure that the data sent is received by the controller? 
Thank you for your help.

Specific HW procedure

`uint8_t u8x8_HW_com_nrf51822(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{

  ret_code_t err_code;
  uint8_t *data;
  switch(msg)
  {

    case U8X8_MSG_BYTE_SEND: {
    //printf("U8X8_MSG_BYTE_SEND\n");
    data = (uint8_t *)arg_ptr;

    APP_ERROR_CHECK(nrf_drv_twi_tx(&twi_instance, 0x3C, &data, 1, false));
            nrf_delay_ms(10);

  //APP_ERROR_CHECK(nrf_drv_spi_transfer(&spi, arg_ptr, arg_int, NULL, 0));

    break;
    }

    case U8X8_MSG_BYTE_INIT: {
     printf("U8X8_MSG_BYTE_INIT\n");

    
        const nrf_drv_twi_config_t twi_config = {
           .scl                = ARDUINO_SCL_PIN,
           .sda                = ARDUINO_SDA_PIN,
           .frequency          = NRF_TWI_FREQ_100K,
           .interrupt_priority = APP_IRQ_PRIORITY_HIGH
        };
    
        err_code = nrf_drv_twi_init(&twi_instance, &twi_config, twi_handler, NULL);
        APP_ERROR_CHECK(err_code);
    
        nrf_drv_twi_enable(&twi_instance);

			
                            break;
        }

      case U8X8_MSG_BYTE_SET_DC: {//Pin Data/Comand

                  nrf_gpio_pin_write(SPIM0_SS_PIN, arg_int);
                          break;
                          }

                         
  default:
  break;
  return 0;
  }
  return 1;

}
`

Gpio_and_delay function
`uint8_t u8g2_nrf_gpio_and_delay_cb(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
switch(msg) {
// Initialize the GPIO and DELAY HAL functions. If the pins for DC and RESET have been
// specified then we define those pins as GPIO outputs.

case U8X8_MSG_DELAY_MILLI:
nrf_delay_ms(arg_int);
break;

case U8X8_MSG_DELAY_10MICRO:
nrf_delay_us(10*arg_int);
break;

  case U8X8_MSG_GPIO_CS:
  nrf_gpio_pin_write(SPIM0_MOSI_PIN, arg_int);				// DC (data/cmd, A0, register select) pin: Output level in arg_int
   
    break;

    case U8X8_MSG_GPIO_RESET:
    nrf_gpio_pin_write(SPIM0_MISO_PIN, arg_int);
     break;

default:
u8x8_SetGPIOResult(u8x8, 1); // default return value
break;
}
return 1;

}
`

olikraus, 2020-01-28:
Maybe you can just query the status byte from the SSD1327 (see the datasheet) to check whether the communication works.

iTzZMaestro, 2020-01-29:
> Maybe you can just query the status byte from the SSD1327 (see the datasheet) to check whether the communication works.

Hello, 
Thank you for your response.
There is communication between SSD1327 and NRF51-DK.
We have looked at the datasheet and we have two questions: what are the variables associated with the value of the R/W bit and the I2C-bus Write data. We would like to check these variables in the debug mode.

olikraus, 2020-01-29:
> what are the variables associated with the value of the R/W bit 

According to the datasheet you can only read the status back (table 9-1)

The source code is also very confusing. What exactly do you want to implement. Is it software emulated I2C?

Then
`u8g2_Setup_ssd1327_i2c_ws_128x128_f(&myScreen, U8G2_R0, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_template);`
will be great and you only need to implement `u8x8_gpio_and_delay_template`.
No other functions are required in this case.

iTzZMaestro, 2020-01-29:
> According to the datasheet you can only read the status back (table 9-1)
>
> The source code is also very confusing. What exactly do you want to implement. Is it software emulated I2C?
>
> Then [...]

Hello,
Thank you for your reply.
I used this setup like you said :

`    u8g2_Setup_ssd1327_i2c_ws_128x128_f(&myScreen, U8G2_R0, u8x8_byte_sw_i2c, u8x8_gpio_and_delay_template);
`
and i have implemented this function 
`uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  ret_code_t err_code1 = 0;

  switch(msg)
  {

  case U8X8_MSG_GPIO_AND_DELAY_INIT: {

    const nrf_drv_twi_config_t twi_config = {
           .scl                = ARDUINO_SCL_PIN,
           .sda                = ARDUINO_SDA_PIN,
           .frequency          = NRF_TWI_FREQ_100K,
           .interrupt_priority = APP_IRQ_PRIORITY_HIGH
        };

        err_code1 = nrf_drv_twi_init(&twi_instance, &twi_config, NULL, NULL);

        APP_ERROR_CHECK(err_code1);
    
        nrf_drv_twi_enable(&twi_instance);

    break; 
    }

  case U8X8_MSG_DELAY_NANO:
  // delay arg_int * 1 nano second
    break;   

  case U8X8_MSG_DELAY_100NANO:
  // delay arg_int * 100 nano seconds
    break;

  case U8X8_MSG_DELAY_10MICRO:
  nrf_delay_us(10*arg_int); // delay arg_int * 10 micro seconds
  break;

  case U8X8_MSG_DELAY_MILLI:
  nrf_delay_ms(arg_int);   // delay arg_int * 1 milli second
  break;

  case U8X8_MSG_DELAY_I2C:
  nrf_delay_us(arg_int<=2?5:1); // arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
  break; // arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us

  case U8X8_MSG_GPIO_CS:
  nrf_gpio_pin_write(ARDUINO_12_PIN, arg_int);	// CS (chip select) pin: Output level in arg_int
  break;

  case U8X8_MSG_GPIO_DC:
  nrf_gpio_pin_write(ARDUINO_10_PIN, arg_int);	// DC (data/cmd, A0, register select) pin: Output level in arg_int
  break;

  case U8X8_MSG_GPIO_RESET:
  nrf_gpio_pin_write(ARDUINO_13_PIN, arg_int); // Reset pin: Output level in arg_int
  break;

  case U8X8_MSG_GPIO_I2C_CLOCK:
  nrf_gpio_pin_write(ARDUINO_SCL_PIN, arg_int);  // arg_int=0: Output low at I2C clock pin
  break; // arg_int=1: Input dir with pullup high for I2C clock pin

  case U8X8_MSG_GPIO_I2C_DATA:
  nrf_gpio_pin_write(ARDUINO_SDA_PIN, arg_int); // arg_int=0: Output low at I2C data pin
  break; // arg_int=1: Input dir with pullup high for I2C data

  case U8X8_MSG_GPIO_MENU_SELECT:
  u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
  break;

  case U8X8_MSG_GPIO_MENU_NEXT:
  u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
  break;

  case U8X8_MSG_GPIO_MENU_PREV:
  u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
  break;

  case U8X8_MSG_GPIO_MENU_HOME:
  u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
  break;

  default:
    u8x8_SetGPIOResult(u8x8, 1); // default return value
    break;
  }
  return 1;
  }
`

Unfortunately, my screen won't turn on. Do you have any suggestions ? 
Thank you for your help.

olikraus, 2020-01-29:
I am running out of ideas

closed 2020-06-14 by olikraus