olikraus/u8g2 #1099: Values in userInterfaceInputValue ()

NorbertNix, 2020-01-26:
Hallo and good evening,
in need some support regarding the above mentioned function in U8G2 Lib.
Using the attached code i have several outputs in the Serial monitor.
To be honest, i dont understand why, To be honest again: im an Java man, and totaly new in C and C++.  
Some thing would be showen after upload.
1. the inital Value is 9 or 7, but never 0.
Turning the Rotary i see the right values at the dispaly but in Serial monitor i get Values like 51,62 or 255 after trigger the Select-Event. What's wrong? The code is addaped from a example i found in your repository. Could you give me an idea what's going wrong. German is possible also. :-)
Attached my sketch:

`#include <RotaryEncoder.h>
#include <Arduino.h>
#include <U8g2lib.h>
#include <Wire.h>

U8G2_SH1106_128X64_NONAME_F_HW_I2C display(U8G2_R0, /* reset=*/U8X8_PIN_NONE); // @suppress("Type cannot be resolved")
int retVal=0;
int inputVal=0;
int *zeiger;
int pos;
RotaryEncoder encoder(4,5);

void setup() {
Serial.begin(9600);
Serial.println("Test Input Value");
  display.begin(/*Select=*/2, /*Right/Next=*/4, /*Left/Prev=*/5, /*Up=*/4, /*Down=*/
  5, /*Home/Cancel=*/A1); // Arduboy DevKit}
display.setFont(u8g2_font_6x12_tr);
}

void loop() {
retVal = display.userInterfaceInputValue("Select Voltage", "DAC= ", *zeiger, 0, 5, 1, " V");
inputVal = *zeiger;
Serial.println(inputVal);
delay(1000);
}`

olikraus, 2020-01-26:
Ich kenne die rotary encoder lib nicht, aber vermutlich ist dein rotary encoder an den pins 4 und 5 angeschossen. Auf der anderen Seite verwendest Du die pins 4 und 5 auch für u8g2. Das kann nicht funktionieren, weil u8g2 pull up buttons erwartet (und eben keinen rotary encoder). Das hat nichts mit C++ vs Java zu tun sondern eher mit hardware entwicklung ;-)

NorbertNix, 2020-01-27:
Hallo Oli,
oh man,,,,,,wie peinlich......, aber trotz dem....Danke.

closed 2020-01-27 by NorbertNix

olikraus, 2020-01-27:
Kein Ding :)