olikraus/u8g2 #1104: u8g2 not working properly on Teensy 4.0

Guzu1987, 2020-02-07:
I am sorry if this is not the right palce to post this.
When using the u8g2 library with a SSD 1309 2.42 inch display and a Teensy 4.0, as soon as the begin() function gets called, horizontal lines appear on the display and do not disappear no matter what I do. 
The clearDisplay() function clears all other objects from the screen, but the horizontal lines remain. Everything else works just fine with the library.

Is there a way to make the oled behave normally? 

I attached a image. (The orientation of the text is not a problem.)
[image]

olikraus, 2020-02-07:
Any documents regarding the OLED? Links? Datasheet?

Guzu1987, 2020-02-07:
> Any documents regarding the OLED? Links? Datasheet?

[Here ](https://www.aliexpress.com/item/32911459164.html)is a link from Alliexpress for the Oled, and [here ](https://www.diymore.cc/products/2-42-inch-12864-oled-display-module-iic-i2c-spi-serial-for-arduino-c51-stm32-green-white-blue-yellow?_pos=11&_sid=9aaf1cabd&_ss=r)is one from the company that makes them, I think.
I am not sure if [this ](https://www.vishay.com/docs/37902/oled128o064dbpp3n00000.pdf)is the correct datasheet for it.

olikraus, 2020-02-07:
One more question: Which u8g2 constructor did you use?

Guzu1987, 2020-02-07:
> One more question: Which u8g2 constructor did you use?

One of the following (I tried all 6 of them) basically what I could find for 1306 and 1309 oleds (HW SPI):

U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 128 bytes]
U8G2_SSD1306_128X64_NONAME_2_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 256 bytes]
U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI(rotation, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]

U8G2_SSD1309_128X64_NONAME0_1_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 128 bytes]
U8G2_SSD1309_128X64_NONAME0_2_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 256 bytes]
U8G2_SSD1309_128X64_NONAME0_F_4W_HW_SPI(rotation, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]

I noticed that the problem appears when the Initialize() function (called in begin() ) is called. Even if I stripped the code to just the initialize() and the power save() functions for the oled, the lines appeared.

olikraus, 2020-02-07:
How many pixel are between each lines? Is the line every 8th pixel?
To be honest, I think the ram of the image is broken.

Guzu1987, 2020-02-07:
> How many pixel are between each lines? Is the line every 8th pixel?
> To be honest, I think the ram of the image is broken.

Yes. The line is every 8th pixel. There are 7 pixels between the lines.

olikraus, 2020-02-08:
Hmm... your last link to the datasheet claims that this is a SSD1306 display.

Guzu1987, 2020-02-08:
> Hmm... your last link to the datasheet claims that this is a SSD1306 display.

On the back of the oled it sais it is SSD1309. I found some more documentation [here](https://drive.google.com/file/d/13Im5G-Qq5yMSgdjmagW6BAxApm9EsB6n/view?usp=drivesdk) and [here](https://drive.google.com/file/d/1asGx-ihNh5kXNCYV3bchUqQuPBEiaYy6/view?usp=drivesdk). [This](https://www.displaymodule.com/products/2-42-dm-oled24-630) is the page where I got the documents from (documentation section).

Guzu1987, 2020-02-08:
Just a little update. After some more trial and error it seems that the library works "fine" (everything works except the horizontal lines appear on the oled) only with the SSD1306 constructor. Any of the SSD1309 constructors make the screen usable only on the bottom 7 pixels - the pixels under the bottom horizontal line. Everything else on the screen remains the same as the first code uploaded - If I modify something in the code and reupload, the top side remains as the first upload of the sketch, and only the bottom side changes - see attached image.

[image]

olikraus, 2020-02-09:
Did you also try the "U8G2_SSD1309_128X64_NONAME2" constructor?

I still think the display is somehow damaged and the horizontal lines is a broken RAM.

Guzu1987, 2020-02-09:
> Did you also try the "U8G2_SSD1309_128X64_NONAME2" constructor?
> 
> I still think the display is somehow damaged and the horizontal lines is a broken RAM.

I will try again the constructor you mentioned and I will let you know if something changes.

I don't think the display is damaged because it works fine with the SparkFun HyperDisplay library. (Exactly the same configuration) No lines appear on the display. Just that the Sparkfun library is very slow and I would like to use your library because is much simpler and faster. 

I also tried your library with another identical oled and Teensy 4.0 and the same horizontal lines appear. Both oleds work fine with the **Arduino Nano** and the Adafruit 1306 library. The Teensy is somehow not working well with the u8g2 library.

olikraus, 2020-02-09:
I understand. The only other thing, what I could imagine is a SPI byte transfer problem.
Actually u8g2 just will use the SPI.h library from the Teensy environment.

Will the display work correctly with Arduino Nano and u8g2? If so, then indeed there might be an issue with the SPI.h lib implementation for Teensy. Maybe also there is a timing issue.

Things to test:
 * Will the SW_SPI constructor work? You have used the HW_SPI constructor. 
 * Will a reduction of the bus speed help (u8g2.setBusClock())?
 * Change the SPI mode 

SPI Modes are 0 to 3 and can be changed by changing the u8g2 code:
Here
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L146
and here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L245

There might be other issues coming from SPI.h. For example the CS handling might be critical: There must be a certain delay after the last bit transfer. But depending on the SPI.h implementation, I can do nothing from u8g2 lib perspective.

Guzu1987, 2020-02-09:
> Did you also try the "U8G2_SSD1309_128X64_NONAME2" constructor?
> 
> I still think the display is somehow damaged and the horizontal lines is a broken RAM.

I tried the constructor. Here is the result:

[Video.zip](https://github.com/olikraus/u8g2/files/4176799/Video.zip)

> Things to test:
>Will the SW_SPI constructor work? You have used the HW_SPI constructor.
>Will a reduction of the bus speed help (u8g2.setBusClock())?
>Change the SPI mode

I will test the options you said this afternoon.
Thank you very much for all your help. Will let you know how the test go.

olikraus, 2020-02-09:
> I tried the constructor. Here is the result:

So it works except for a 32 pixel shift, correct?

To fix the 32 pixel shift, try to change the start line:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L167
Instead of 0x40 use 0x60. Will this help?

Guzu1987, 2020-02-09:
> > I tried the constructor. Here is the result:
> 
> So it works except for a 32 pixel shift, correct?
> 
> To fix the 32 pixel shift, try to change the start line:
> https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L167
> Instead of 0x40 use 0x60. Will this help?

I changed that line. Same result:
[image]

Guzu1987, 2020-02-09:
> * Will a reduction of the bus speed help ([u8g2.setBusClock()](https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock))?

I changed the bus speed but the oled still displays the horizontal line. 
The change in speed is evident from the refresh rate on the screen. Still the lines remain.

olikraus, 2020-02-09:
> I changed that line. Same result:

It depends on the constructor. I mean, the line in the code might be different depending on the constructor  you use.

> I changed the bus speed but the oled still displays the horizontal line.

What about the software SPI constructors?

Guzu1987, 2020-02-10:
> What about the software SPI constructors?

I just tried to use SW SPI and the **display works** with the **SSD 1306** constructor. With the SSD 1309 constructor the display is shifted. See video attached. 

However, I would really like to use HW SPI. Could a fix be implemented? Or what could I do in order to make the oled work in HW SPI?

[Video.zip](https://github.com/olikraus/u8g2/files/4181103/Video.zip)

olikraus, 2020-02-10:
Ok, so if I understand correctly, there are two problems which are independent:

- SW SPI vs. HW SPI
- SSD1306 (works correctly) vs SSD 1309 (display shifted)

I might solve the second point, but the first point seems to be a problem with your Teensy board.
Teensy comes with a hardware abstraction layer, which includes the SPI interface. If SW SPI works, but HW SPI not, then Teensy Software needs to be modified. For any other board SW SPI and HW SPI both can be used.

Guzu1987, 2020-02-10:
> Ok, so if I understand correctly, there are two problems which are independent:
>
> * SW SPI vs. HW SPI
> * SSD1306 (works correctly) vs SSD 1309 (display shifted)
>
> I might solve the second point, but [...]

I understand. So my Teensy board is deffective? Or the SPI.h file needs to be rewritten somehow for the Teensy?

olikraus, 2020-02-10:
I am sure the Teensy board is fine, but I assume, the low level software has an issue.
The SPI support for the Teensy 4.0 is probably very new and maybe there is an issue: https://github.com/PaulStoffregen/SPI/blob/master/SPI.cpp#L1264

The other problem could be this: Maybe the Teensy 4.0 is simply too fast.
As I mentioned earlier, you could try to lower the busClock with HW SPI. 
If you want to test a specific constructor you could also tweak the delays. Especially the last bit to CS might be to quick.
For the SSD1306 noname constructor this delay is here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c#L294

Guzu1987, 2020-02-11:
> I am sure the Teensy board is fine, but I assume, the low level software has an issue.
> The SPI support for the Teensy 4.0 is probably very new and maybe there is an issue: [...]

Thank you very much for all the information. And I really appreciate that you took the time to try to solve the issue.
I will try the things you suggested and I will post the results if I have any positive ones.

Guzu1987, 2020-02-11:
I tried changing the Bus speed again using the **u8g2.setBusClock()** but it seems to have no effect on the oled. The speed of the drawings does not change. I use the U8G2_SSD1306_128X64_NONAME_F_4W_HW_SPI constructor. I placed the function call right after the u8g2.begin(). I tried even u8g2.setBusClock(1000) and no effect. Am I doing something wrong here?

Also, I tried changing the delays in the file you specified. That had no effect on the horizontal lines. 

I think I will try to open an issue on the SPI library developed by PaulStoffregen. Maybe he knows what is going on.

Could you tell me if there is a way to make the SW SPI work faster? I saw there was an issue here and a user comments out some delays n the files, but those delays are already commented out in the library I have. Thanks.

olikraus, 2020-02-11:
There are ways to make sw spi faster, but this is no portable code and depends on the target controller/board.

Guzu1987, 2020-02-11:
> * Change the SPI mode
> 
> SPI Modes are 0 to 3 and can be modified by changing the u8g2 code:
> Here
> https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L146
> and here:
> https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L245

I changed the code as you mentioned and IT WORKS! I changed the spi_mode = 3 here:
https://github.com/olikraus/U8g2_Arduino/blob/master/src/clib/u8x8_d_ssd1306_128x64_noname.c#L300
**Thank you very much**.
What did this setting change? Just to know what is been eating my days 

I am sorry that is was such a simple solution and I did not know how to implement it.
I will close the issue.

Thank you very very much for the information.

closed 2020-02-11 by Guzu1987

olikraus, 2020-02-11:
Well... Yes, but it means, that there's an issue with the spi implementation.

Guzu1987, 2020-02-11:
> Well... Yes, but it means, that there's an issue with the spi implementation.

I understand. Well, it seems that the guys that produced the Teensy bought some of these displays and will work on a fix.