olikraus/u8g2 #1105: Problem Display 64128COG - Driver KS0713

Borxo, 2020-02-08:
Hi, I have a problem with this display adapter to the library.
I am connecting the display via SPI, with the "ST7565 64128N" constructor. I get nothing, not even to backlight.

Conexiones: 

Arduino mega 2560 
Reset= D8 
DC=D9 
CS=D10 
SID=D51 
SCLK=D52 

Thank you

olikraus, 2020-02-09:
Maybe the display power lines are missing?

Borxo, 2020-02-09:
The connections I am using are those defined in your datasheet. I have tried both Serial and Parallel conumication.
[image]
[image]
I also leave the datasheet of the manufacturer
`https://docs.rs-online.com/4ac5/0900766b806dda2c.pdf`

Thank you so much for everything

olikraus, 2020-02-09:
The idea is this: You need to check your wiring. Sending datasheet diagrams does not really help here.

For example you mention, that Reset is connected to D8. This information alone will not help to validate your setup. You also need to mention where you have connected D8. So a proper connection would at least be this:
U8g2 Constructor Argument = Arduino Pin = Display Pin
Reset = D8 = RST

Maybe you can provide this connection for the remaining pins (just to check, that all is planned corretly). Then you could make a screenshot of your wiring and we can crosscheck your wiring together.

Borxo, 2020-02-09:
I am using an Arduino Mega 2560, the connections with it are as follows.
[image]

The constructor that I am using is:
`U8G2_ST7565_64128N_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 52, /* data=*/ 51, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`
I use pins 52 and 51, since that is where the MOSI and SCK connections are.
I am using a PCB designed by me to have the capacitors already on.
[image]

olikraus, 2020-02-09:
> I am using a PCB designed by me to have the capacitors already on.

Nice, but if you want me to support here, you need to tell me how the pins from your display are connected?

For example: To which Arduino Mega pin is RS connected?
Please complete the table from above.

Borxo, 2020-02-09:
if I haven't understood it wrong, these are the Pin's should connect.

|Pin Display | Pin Arduino Mega |
 -----------------------------------
 |     CS1B    |            Pin 10          |
 -----------------------------------
 |      RST     |            Pin 8            |
 -----------------------------------
 |   RS (DC)  |            Pin 9            |
 -----------------------------------
 | SID(DB7)  |            Pin 51          |
 -----------------------------------
 | SCLK(DB6)|            Pin 52         |
 -----------------------------------
 |       MI       |            VCC            |
 -----------------------------------
 |       PS       |            GND            |
 -----------------------------------
 |      VCC       |            3.3V           |
 -----------------------------------
 |      GND       |            GND         |

Thank you

olikraus, 2020-02-09:
Looks good from u8g2 perspective. Is your pcb correct? How is the PCB connected to Arduino?

Borxo, 2020-02-09:
I have the capacitors connected as in the Datasheet, and the other connections are made by means of wire to the arduino pins.
The only thing that the PCB does is to extract the pins from the Display.

olikraus, 2020-02-09:
> I have the capacitors connected as in the Datasheet, and the other connections are made by means of wire to the arduino pins.

Hmm.. you always tell me that you did everything correctly. I wonder why you need my help.
Since you did not provide enough information (and the information what is there is there only because I asked for it), I can NOT tell you what you made wrong. 
So the only thing what I did, is to ask questions about what could be wrong. If you only reaction is, that you did everything correctly, then I also do not know what to do...

So this is my last suggestion: Make a list of things what you could be wrong. This list can be easily 20 to 30 line items. Do not think that things are correct. You need to prove that all thinks are correct.

One question could be this: Did I use the correct capacitors? Is the type correct? is value correct?
Do not answer this. My advice is: Search for such questions and answer these questions for yourself.
(BTW: You have indeed selected the wrong type of caps)

Borxo, 2020-02-09:
It was not my intention to offend, only when I say that everything is correct, I followed the datasheet scheme for PCB creation.

I know something must fail, that's why I'm consulting you. I assume that I am not as expert as you are on the subject, so I want to be advised by you.

All the information I have right now about it, is what I have currently provided. He just wanted to know if he could come up with a solution.

It was also to know if the constructor that I am using was suitable for said screen, since I am using one that suits the driver that has said display.

Borxo, 2020-02-09:
I have just assembled the circuit again and I have been measuring with a polymeter the voltage that reaches the LCD, I have observed that when I have removed the VCC cable, it continues to reach the 3.3V LCD.

olikraus, 2020-02-09:
> I am connecting the display via SPI, with the "ST7565 64128N" constructor. I get nothing, not even to backlight.

Are you sure there is a backlight? There might be no backlight. It sometimes has to be mounted separatly.

> It was also to know if the constructor that I am using was suitable for said screen, since I am using one that suits the driver that has said display.

Try the contrast demo of u8g2. It will loop over all possible contrast value. It might be the case, that you display requires a different contrast: I mean, the default contrast value may not fit for your display.

Borxo, 2020-02-09:
Right now I have checked what he has told me and what I have discovered has killed me.

In the Pin Vout that is connected to the capacitors, if I am not mistaken it is for the power supply of the screen. This voltage must be between 9V to 11V. Sometimes if you get that voltage but another stays in the circuit input voltage and the screen goes black but most of the time it is blank.

Could it be that some capacitor is not working properly?
As I have read the dampers serve to raise the input voltage to feed the screen, right? or am I wrong?

closed 2020-02-24 by Borxo

olikraus, 2020-02-25:
Oh, i missed your last reply. 
You should use ceramic caps between 0.1uF and 1uF.