olikraus/u8g2 #1108: SSD1327 i2c

Mael19, 2020-02-13:
Hello, I have struggle setting up my ssd1327. I successfully managed to use other screens but this one doesn’t seem to work.

I’m starting to think that it’s broken, but when I run an i2c scanner it is recognised...

How can I know whether it’s broken or not supported by this library ?

My model : https://a.aliexpress.com/_Tzjet

olikraus, 2020-02-14:
I assume there is some wiring problem. Which constructor did you use? Any picture of the setup?

Mael19, 2020-02-14:
Well I tied with all the constructor that seemed to fit my screen, so the followings :
//U8G2_SSD1327_EA_W128128_1_SW_I2C u8g2(U8G2_R0, 22,21, U8X8_PIN_NONE);
//U8G2_SSD1327_EA_W128128_1_HW_I2C u8g2(U8G2_R0,U8X8_PIN_NONE);
//U8G2_SSD1327_MIDAS_128X128_1_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);

Where 22 and 21 are my hardware i2c pins, it is wired properly and I can make other devices work fine.

Here is also a picture of the setup, as you can see there's nothing very complicated.
[image]

olikraus, 2020-02-14:
Hmmm These Profiboards are not very reliable. Maybe shift everything by one row.

Mael19, 2020-02-15:
I already tired, and I rewired the setup like 10 times. Somebody on another forum suggested using pull-up resistors, which I tried but nothing seems to change.

olikraus, 2020-02-15:
Difficult to say what is wrong here. If this would be in my lab, I would start to analyse the I2C traffic on the bus.

Mael19, 2020-02-17:
Well I don't have anything to analyse the i2c traffic... I notices that when using the following config some lines just blink a few ms when resetting the board. I don't know if that means anything ?

olikraus, 2020-02-17:
Board reset? Then something else might be wrong.

Mael19, 2020-02-17:
Yes, either after a board reset (the ESP32) or when the script is done transferring / the board is powered on. Also I can hear a weak buzzing from the screen board

olikraus, 2020-02-18:
Sounds like a broken display electronics.

Mael19, 2020-02-19:
Well let's assume it's broken, and I don't have the time or the knowledge to try to prepare it. Thanks for your time !

closed 2020-02-19 by Mael19