olikraus/u8g2 #1109: Arduino Nano BLE 33 sense + OLED SSD1327 128x128 SPI

GitLorenzo, 2020-02-14:
Hello,
I have used an OLED SSD1327 128X128 pixels (from waveshare) together with an Arduino Nano (the old model) and the amazing U8G2 library. It worked perfectly. Recently I bought an Arduino Nano BLE 33 sense, thinking that I could have easily substituted it without changing the software and the wiring, but it doesn't work.

As an example:

- I tried the U8G2 example  > page buffer > hello world.
- I uncommented the constructor line:
`U8G2_SSD1327_MIDAS_128X128_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

This works with the old Arduino Nano, but does not work with the BLE 33 sense.

Here the connections I made OLED > Nano BLE 33 sense:

VCC > 3.3volt (between pin REF and D13)
GND > GND
DIN > D11
CLK > D13
CS > D10
DC > D9
RST > D8

Do you have any idea about how to solve this issue?

Thank you very much in advance :)

Best,
Lorenzo

olikraus, 2020-02-14:
Maybe there is a conflict with other sensors on the same bus. Try HW SPI instead.

GitLorenzo, 2020-02-14:
Dear olikraus,
thank you for your comment. I tried HW SPI but does not work...

olikraus, 2020-02-14:
So you tried this constructor?
```
U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
```

GitLorenzo, 2020-02-14:
Exactly,
that's what I tried... but does not work.

olikraus, 2020-02-15:
I do not know much about the ble board. Maybe the display expects 5V but the new board only has 3.3 volt output.

GitLorenzo, 2020-02-15:
But the tech specs of the oled say that works at 3.3 volts too

olikraus, 2020-02-15:
Ah, ok. Actually I don't know what could be the reason. May use complete different gpios with sw spi

closed 2020-06-14 by olikraus

amaussa, 2021-04-18:
see https://forum.arduino.cc/t/arduino-nano-ble-33-sense-oled-ssd1327-128x128-spi/636461